grbl/grbl #1155: create loop depend on G92

MohamadAlsadi, 2017-01-05:
how I can create loop to repeat group of G-code commands by using G92 ?
 I have this code 

g0 x0 y0
g2 x-19.05 y-19.05 i-19.05 j0
g3 x-38.1 y-38.1 i0 j-19.05
g2 x-57.15 y-57.15 i-19.05 j0
g3 x-57.15 y-95.25 i0 j-19.05
g2 x-38.1 y-114.3 i0 j-19.05
g3 x-19.05 y-133.35 i19.05 j0
g2 x0 y-152.4 i0 j-19.05
g4 p1
g10 l20 y0
 
I want repeat my code 5 times depend on G92

electrokean, 2017-01-05:
You'll need to do this in your CAM post processor or GUI g-code sender

MohamadAlsadi, 2017-01-05:
@electrokean 
no my dear, this programs generate commands not supported in GRBL v0.9

electrokean, 2017-01-05:
Well grbl doesn't support loops or macros, and probably never will.
It sounds like you will just need to open your g-code in a text editor and copy/paste it multiple times with your work coordinate offset change in between each part.

MohamadAlsadi, 2017-01-05:
@electrokean 
I think that g92 may be solve this proplem