grbl/grbl #1156: 8bit resolution variable Spindle

adibiut, 2017-01-06:
Hi everybody. My CAM software for Laser engraving, generates 0-255 spindle speed, but grbl has a 0-1000 range. which part of code should be changed in to have a 8bit Spindle enable output?

chamnit, 2017-01-06:
@adibiut : Grbl maps 0-1000 rpm to 8-bits. The rpm values are arbitrary and only there to associate an rpm to voltage.

closed 2017-01-06 by chamnit

adibiut, 2017-01-06:
I know that. I want to have a 5v voltage when S255 and 0v voltage where S0
but current version sends out 5v when S1000

langwadt, 2017-01-06:
change the max_rpm setting to 255, I think it is $30

adibiut, 2017-01-06:
I'm using 0.9j and there is not such setting

chamnit, 2017-01-06:
@adibiut : It can be changed in the config.h file. Otherwise use v1.1, which has $30 available as an rpm max setting.

mayhem2408, 2017-01-07:
If you want to change the default range to be between 0-255 instead of 0-1000, you can change the defaults.h file and modify DEFAULT_SPINDLE_RPM_MAX which will be 1000 by default.