grbl/grbl #1157: New to CNC

scolbert43, 2017-01-09:
Hello, hopefully someone is willing to help me.  I recently bought a CNC, finished putting it together, and now ready to design something.  This is what it says for me to use, Designed to use GRBL firmware on the Arduino Uno. But there is where I am stuck, where should I go, what should I do next, what am I missing.  Hoping someone can understand how confusing this is for me and able to give me step by step information to help.   Thank you in advance for your help!!

Scott

jahnj0584, 2017-01-09:
You design your project in CAD, like fusion360 or Sketchup and create the
toolpaths in a CAM software, like sketchucam or fusion 360.

whitetd, 2017-01-09:
First, do you have the uno with grbl firmware?

Norwegian-Gadgetman, 2017-01-09:
Hi!

I find that the ShapeOko community is a good place to start.
First, read the relevant parts of the Wiki:
<http://www.shapeoko.com/wiki/index.php/Workflow>
Then check out the forum.
<http://www.shapeoko.com/forum/>

:-)

Trygve

scolbert43, 2017-01-09:
Please excuse me for being not educated on this, whitetd: what is uno?  Is it what comes with the CNC or something I need to download?

whitetd, 2017-01-09:
Did you get a kit?,

mayhem2408, 2017-01-09:
@scolbert43 An Uno refers to an Arduino Uno. It is the hardware the GRBL firmware is install on. The Uno can them control your stepper motor drivers, spindle controllers and laser drivers. The Uno becomes your link between your PC and the CNC Rig.

scolbert43, 2017-01-09:
Yes, it came with a Arduino Uno.