grbl/grbl #1158: PostP question

milezdr, 2017-01-10:
Hi to all.Does we need new postprocesor for the cam software with the new version of grbl 1.1 I'm using Vectric aspire ?
Thanks

X3msnake, 2017-01-11:
I think aspire has a built in post processor for GRBL. You choose it when
you export the tool.

Can you confirm?

X3msnake, 2017-01-11:
Sorry misread your question.

Not being able to confirm just now I'll say you will not need a newsletter
post processor.

Here is the list of v1.1 supported gcodes at the moment.

   - Multiple full circle arcs with G2 and G3 arcs with a P word is not
   supported.
   -

   Laser mode alters the operation of M3, M4, and spindle speed S word
   changes. See the Laser Mode
   <https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode> page for
   details.
   -

   Supported G-Codes in v1.1
   - G0, G1: *Linear Motions*
      - G2, G3: *Arc and Helical Motions*
      - G4: *Dwell*
      - G10 L2, G10 L20: *Set Work Coordinate Offsets*
      - G17, G18, G19: *Plane Selection*
      - G20, G21: *Units*
      - G28, G30: *Go to Pre-Defined Position*
      - G28.1, G30.1: *Set Pre-Defined Position*
      - G38.2: *Probing*
      - G38.3, G38.4, G38.5: *Probing*
      - G40: *Cutter Radius Compensation Modes*
      - G43.1, G49: *Dynamic Tool Length Offsets*
      - G53: *Move in Absolute Coordinates*
      - G54, G55, G56, G57, G58, G59: *Work Coordinate Systems*
      - G61: *Path Control Modes*
      - G80: *Motion Mode Cancel*
      - G90, G91: *Distance Modes*
      - G91.1: *Arc IJK Distance Modes*
      - G92: *Coordinate Offset*
      - G92.1: *Clear Coordinate System Offsets*
      - G93, G94: *Feedrate Modes*
      - M0, M2, M30: *Program Pause and End*
      - M3, M4, M5: *Spindle Control*
      - M8, M9: *Coolant Control*

milezdr, 2017-01-11:
Yes in aspire it have Grbl post procesor but for Grbl 0.8 version so that's why I was wondering if I need to do some changes so I can use the new grbl 1.1 firmware or the old processor is god? 
Thanks in advance

jahnj0584, 2017-01-11:
The old post will be fine. I used another post and it worked fine except
for the T02 command at the beggining. UGS allows to skip the commands that
aren't supported

closed 2017-01-12 by milezdr

X3msnake, 2017-01-12:
T02 is the tool select

if you have problems with this you can open the post Processor, search for
T remove it and save it.

+---------------------------------------------------
+  Commands output at the start of the file
+---------------------------------------------------

begin HEADER

"T1"
"G17"
"G21"
"G0[ZH]"
"G0[XH][YH][S]M3"

X3msnake, 2017-01-12:
Also if you want to use Fusion 360 here is a better GRBL post Processor

https://github.com/Strooom/GRBL-Post-Processor

2017-01-12 12:39 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> T02 is the tool select
>
> if you have problems with this you can open the post Processor, search for
> T remove it and save it.
>
> +---------------------------------------------------
> +  Commands output at the start of the file
> +---------------------------------------------------
>
> begin HEADER
>
> "T1"
> "G17"
> "G21"
> "G0[ZH]"
> "G0[XH][YH][S]M3"
>
>
>
>
>> Thanks
>>
>> <https://github.com/grbl/grbl/issues/1158#issuecomment-272143304>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva