grbl/grbl #1161: Vote for GRBL post update in fusion 360 to include new v1.1 features

tklus, 2017-01-13:
please vote this up guys so we can get the post team to make the changes to support GRBL v1.1 laser modes.

http://forums.autodesk.com/t5/hsm-ideas/update-grbl-post-to-allow-laser-program-posting/idi-p/6795471

109JB, 2017-01-13:
Why duplicate this request when you already posted it on the V1.1 site and this site isn't for V1.1 anyway?

tklus, 2017-01-13:
Just trying to spread the word. Some folks may not be subscribed to the new grbl git hub location

X3msnake, 2017-01-13:
adding that is probably just a matter of changing the Post Processor since
F360 already has the laser function implemented

I can't test it since i don't have a GRBL laser but i can check what
changes need to be made in order to post correctly if noone has yet done
this

tklus, 2017-01-13:
That would be great! Attached is the stock GRBL post and the Smoothie post. I don't know much at all about post editing. if you get the GRBL post modified I would be happy to test it out.

[grbl.zip](https://github.com/grbl/grbl/files/705577/grbl.zip)

X3msnake, 2017-01-14:
what's the difference in the Laser Program

Is it only the M5/M3 commands?

do you have a working grbl program for your laser that you can share

tklus, 2017-01-14:
I can when I get home. So one of the issues is when I try and post a laser tool path in fusion with the grbl post, fusion complains that the post does not support laser cutting.

Also, the smoothie post allows you to set the power in the post options when posting a program. This would be a nice feature to have in the grbl post. I attached a screen shot of what I am describing in the autodesk idea post linked above.

Hopefully that makes sense.

Tim

X3msnake, 2017-01-14:
Sure we can use smoothie as a base the code output is pretty clean and I
believe it can be use almost out of the box with GRBL except for the M
codes that i Think are not supported.

drop here a link of a simple grbl program that you have working on your
laser machine like cutting a square and a circle and i'll have a look

tklus, 2017-01-14:
here is a simple laser program that is 2 squares cut out. I think I output this with the smoothie post but I had to edit the program to make it run. for some reason the speeds were not outputting for the G1 cut moves. it just output an F0 command. some of these codes may not be needed. I usually leave out the G28 aux home commands

let me know if you need anything else.

Thanks a ton for working on this!

Tim

%
; grid
; T0 D=0 CR=0 - ZMIN=0 - laser cutter
G90
G17
G20
G28 G91 Z0
G90

; Contour1
T1
M107
M3
G0 X-0.3196 Y-0.3105 F100
G0 Z0
G1 X-0.3596 F15 S255
G3 X-0.3996 Y-0.3505 J-0.04
G1 X0.6101
G1 Y0.362
G1 X-0.3996
G1 Y-0.3505
G3 X-0.3596 Y-0.3105 J0.04
G1 Y-0.2705
G1 X-1.5 Y-1.0405 F100 S0
G1 Y-1.0005 F10 S255
G1 X-1.5005 Y1
G1 X1.5 Y1.0005
G1 X1.5005 Y-1
G1 X-1.5 Y-1.0005
G1 X-1.54
M5
M107
M5
G28 G91 Z0
G28 X0 Y0
%

X3msnake, 2017-01-14:
This code you posted is working on GRBL laser machine right?

You are able to cut with this code.

Also can you share the f360 file that you used.