grbl/grbl #1164: Change frequency of PWM

adibiut, 2017-01-21:
Hi everybody.
As I know, PWM frequency for variable spindle in v0.9 is 7.8 kHz . How Can I change it to 1kHz? which part of code should be changed?
(I know in grbl 1.1 it has been 1kHz. but I want to use v0.9)

chamnit, 2017-01-21:
@adibiut : Look at the cpu_map.h file in v1.1. There are a few lines showing you have to alter the PWM frequency. Just do the same in the v0.9 cpu_map file.

closed 2017-01-21 by chamnit

adibiut, 2017-01-21:
There isn't something like this in v0.9 :  #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98k

I'm not a pro. Would you please tell me exactly what I should change? or send the modified cpu_map.h file to my email? That would be very nice of you.
adib.z1990@gmail.com

adibiut, 2017-01-23:
Why do you close it?

adibiut, 2017-01-24:
?

electrokean, 2017-01-25:
It would have been closed because there is no intention to make changes here, and this is not a general support forum.  You have full access to the source so you can make your own changes.
I think you want to look at spindle_control.c:94 for the variable speed prescaler configuration.