grbl/grbl #1169: How to controll differnet voltage stepper motors

Erdc, 2017-01-30:
I am making a PNP CNC, and have the X & Y axis running on 24V from a GRBL shield, but the rotating vacuum nozzle stepper motor that I have requires just 3.3v and my Z axis requires 5V. How do I go about wiring them all in so that the  OPenPNP software can interact with each one?

109JB, 2017-01-31:
Are the voltages that you are referring to are the rated voltages of the motors?

If so, then just ignore them. Those voltages are the maximum continuous voltage that will not exceed the current rating of the motor.  However, the stepper drivers used are "chopper" drives and regulate the current by chopping the current to keep it in check.  In short the "rated" voltage on the motor is somewhat meaningless for our purposes and 24V will be fine as long as you adjust the drivers to supply only the rated current of the motors.

electrokean, 2017-01-31:
I'm guessing you mean that one axis has its own external driver that takes logic level signals.  In that case, you just don't plug in a stepper driver module into that axis slot, and you can tap off the logic level signals from the appropriate spots.
In fact there are pass through modules for this - e.g. the Protoneer ones http://www.ebay.com/itm/Pololu-Socket-To-External-Driver-conversion-board-/272511745429

Erdc, 2017-01-31:
The image below is the driver that I am using, 
109JB, you are correct, I am talking about the rated voltages of the motors. Thanks for the clarification on voltage ratings.

electrokean, I will look into that pass through module.

[image]

electrokean, 2017-01-31:
@erdc So just to clarify, the GRBL shields are designed to run all stepper drivers plugged into the shield from the same motor supply voltage.  I was just thinking that one way to do what you want would be to cut or bend the VMOT pin on some of the stepper driver modules so it doesn't go into the socket, and wire those pins to the alternate motor power supply voltage.  Or you use something like the pass through module, and separate stepper drivers.  Hope that makes sense.