grbl/grbl #1171: Mirror axis

sigifre, 2017-02-01:
There are any way to mirror the Y step pulse to another Arduino pin for  2 motors in the same axis? (having one as slave)

chamnit, 2017-02-01:
Connect your stepper driver to the same pin.

closed 2017-02-01 by chamnit

sigifre, 2017-02-01:
that does not work good...
In other platforms they use slaves for that reason, they use axis A and u can configure it as slave of any other.

vMeph, 2017-02-01:
Just put in paralel the enable, dir and step on the drivers and will work

109JB, 2017-02-01:
As said, just wire the drivers in parallel and if one if the motors turns the wrong direction the just switch the wires for one coil going to the motor. For example move A+ to A- and vice versa.

sigifre, 2017-02-01:
it is in that way but the max speed is reduced, so in a moderate speed it works good. I just want to try a mirror axis to see if the speed can be higher.

jahnj0584, 2017-02-01:
The driver's decide that, not grbl. I can do 30m/min with two Y motors

electrokean, 2017-02-02:
As said above you can definitely slave an axis just by paralleling the step & direction outputs.
Many drivers have opto-isolated inputs, and it may be possible that paralleling driver inputs could put too much load on the ATmega328 output pins affecting signal integrity and limiting may pulse rate.  In this case it may be necessary to insert some signal buffering.

sigifre, 2017-02-02:
Thats why Im looking another pulse signal pin, it seems that the only way is modifying the library.