grbl/grbl #1173: Core XY mechanism not working

sachinsachi, 2017-02-03:
Hello! I am new to CNC programming and I am trying to build a core XY mechanism using grbl firmware.. I am using uno CNC shield and I uploaded grbl firmware version 0.9. I am using universal g code sender to test my guide and it's giving me issues. When I try giving x and y commands it's working but I am unable to get the expected shape when I upload my gcode.

mayhem2408, 2017-02-03:
@sachinsachi Have you compiled GRBL to use CoreXY. If you are starting a new project, I suggest using GRBL V1.1. I run a CoreXY rig and V1.1 and it works great. What shape are you trying to create and what is the result?

sachinsachi, 2017-02-04:
I am trying to create circular shapes mostly like spiral etc. And even homing is an issue .

electrokean, 2017-02-04:
As asked by @mayhem2408 Have you compiled GRBL to use CoreXY?  It is not the default.
Also, please look at the newer version of grbl 1.1 at https://github.com/gnea/grbl

sachinsachi, 2017-02-06:
Hi @mayhem, @electrokean yes I have compiled grbl 1.1 and set it up for Corexy. Still the problem persists. U haven't connected the limit switches yet and the gcode is not behaving as expected.

sachinsachi, 2017-02-06:
It works fine if I try to run x and y in the universal gcode sender

sachinsachi, 2017-02-06:
But when i  try uploading my gcode it give me junk results

chamnit, 2017-02-06:
@sachinsachi : You are going to have to be more descriptive of what's wrong. Just saying "it give me junk results" doesn't help us help you. First, are your motors arranged and ROTATE the way described in the CoreXY [website](http://www.corexy.com/theory.html)? Otherwise your machine will move in the wrong directions.

sachinsachi, 2017-02-06:
Hi @chamnit yes I have arranged my motors as per the Corexy mechanism.. I am using arduino uno board with uno CNC shield v3. I have modified the code to work for Corexy and as of now for testing purpose I haven't connected my limit switches. When I open the Universal gcode sender and try running x and y motors it works fine. If I upload a gcode file containing curved shape , it gives me junk results

chamnit, 2017-02-06:
You have to do better than that. "Gives me junk results" can mean anything.