grbl/grbl #1175: Machine status still shows zero?

Nermin1985, 2017-02-03:
Hi, I'm Nermin made my first CNC machine that uses an Arduino Uno Grbl 1.1 and easy driver v4.4
The problem that I have is Machine status, always showing me zero for x, y, z osu.Motori are on the move or the status of the machine that is the position of the machine shows me that curled at zero.

Here's a link to the results: https://www.youtube.com/watch?v=ettBJoax04Q

Do you can help to solve this problem?

Nermin1985, 2017-02-03:
Do you can help to solve this problem?

adibiut, 2017-02-03:
use new version of UniversalGcodeSender (2.0)
although it's still unstable

electrokean, 2017-02-03:
Yes, I think the old version of UGS is not compatible with the updated grbl 1.1 protocol.
So it is not a problem with grbl, but with UGS.
Have you tried bCNC or GRBLpanel?

Nermin1985, 2017-02-03:
I tried to Grbl Controller 3.6.1 and the same problem ...
Is there a version UniversalGcodeSender (2.0)? Try to find, but to no avail if there is a link?

Nermin1985, 2017-02-03:
I tried to GRBL panel and operating normally, showing exactly cordinate ...

I wish I could work with UniversalGcodeSender or
Grbl Controller

If you have a link or a solution, I'll be very grateful;)

adibiut, 2017-02-03:
http://bit.ly/1hftIhy

vMeph, 2017-02-03:
UGS platform?

electrokean, 2017-02-04:
@Nermin1985 glad you could confirm it works with GRBL panel.  So as I mentioned this isn't a problem with grbl, but with some sender applications not supporting the new status reporting of grbl 1.1
I suggest you check out the repositories/forums for your preferred sender application, and if necessary report the issue to the maintainer of those apps.

@adibiut it isn't a good idea to post shortened links to files from unknown sources.  Much better to refer the OP to the actual source of those file.

electrokean, 2017-02-04:
For UGS see http://winder.github.io/ugs_website/#grbl-11-features
Regarding GRBL Controller, see https://github.com/gnea/grbl/issues/117

electrokean, 2017-02-04:
Also, as you are using grbl 1.1 you should be reporting any issues to the 1.1 repo - this repo is for grbl 0.9
https://github.com/gnea/grbl

Nermin1985, 2017-02-04:
**Yes now I see that I was wrong in reporting;)
Thank you very much for your help and a lot of you have helped me, Please understand I am new to Grbl and CNC machines.
And the English language is not praised for me;)**