grbl/grbl #1176: GRBL locking up

Russellfam, 2017-02-04:
I have a section of gcode I'm running on grbl and its being sent by UGS. It's locking up intermittently for no apparent reason, no error message returned to UGS. UGS looks like its waiting for the next command.

The lockup happens at different times and locations. I'm cutting a profile and it sometimes stops part way through the first cut, other times it will complete the first cut and then drop down the z by 4mm and stop halfway through the second cut. It's nowhere near any of the limits. I have hard limits off.

It looks like the Arduino locks up and I have to unplug USB and reconnect to get it to respond to USG. Closing and opening USG connection does not work. 

I've done other profiles almost identical several times over the last day with no issues. I've attached the gcode file generated by CamBam
[Custom_outline_mirror.txt](https://github.com/grbl/grbl/files/752361/Custom_outline_mirror.txt)

chamnit, 2017-02-04:
@Russellfam: Please post your $$ Grbl settings and $I build info string. Problems like these are usually related to settings. Or if you are using a knockoff clone Arduino that has a questionable USB serial chip.

Russellfam, 2017-02-04:
$I
[VER:1.1e.20170114:]
[OPT:V]

$$

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 7    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 600.000    (Homing search seek rate, mm/min)
$26 = 244    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 1280.000    (X-axis travel resolution, step/mm)
$101 = 1280.000    (Y-axis travel resolution, step/mm)
$102 = 1600.000    (Z-axis travel resolution, step/mm)
$110 = 650.000    (X-axis maximum rate, mm/min)
$111 = 800.000    (Y-axis maximum rate, mm/min)
$112 = 400.000    (Z-axis maximum rate, mm/min)
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 30.000    (Z-axis acceleration, mm/sec^2)
$130 = 865.000    (X-axis maximum travel, millimeters)
$131 = 575.000    (Y-axis maximum travel, millimeters)
$132 = 100.000    (Z-axis maximum travel, millimeters)

chamnit, 2017-02-04:
@Russellfam : Nothing too unusual. A few more questions before I break out the test rig. How are you flashing Grbl? If you are compiling, tell me your Arduino IDE version. Is your Arduino a genuine Arduino Uno? If not, what is it and what is the USB serial chip on it. Should be labeled a atmega16U or FTDI or CH380.

electrokean, 2017-02-04:
My guess would be electrical noise from the spindle coupling through your USB cable and causing a disconnect.
Has this only just started happening?  If so have you moved any of your cables around?
Do you have any noise filtering, in particular a ferrite on your USB cable?  If not, dig around and see if you have a USB cable with a ferrite.  The ferrite looks like a lump on the cable.  You can also buy clip on ferrites from various electronics suppliers.
See https://www.google.com/search?q=ferrite+usb+cable&tbm=isch for sample images

Russellfam, 2017-02-04:
Hey, thanks for the replies. 
I flashed it from the hex file using Xloader
The board is an arduino uno Funduino and the serial is the WEGA16U2

Russellfam, 2017-02-04:
I've recently upgraded the GRBL and the UGS. Since then I've cut several similr profiles with no issues. this problem started this afternoon and I've changed nothing on the system.

chamnit, 2017-02-04:
@Russellfam : I've been running your job on two different Arduinos with the same firmware build and your settings (with soft limits turned off). I'm streaming the program with the stream.py script in the Grbl repo to determine if it's GUI related. So after 3 runs each for a total of 6, no locking up so far. 

Grbl v1.1's interface was updated to contain more realtime information, so

chamnit, 2017-02-04:
Whoops.. Accidental click. 

Grbl v1.1's interface was updated to contain more realtime information, so it might be a GUI bug. I'll keep running the job and will report a lock-up, if I run across one.

Russellfam, 2017-02-04:
thanks for the help...sure hope we can solve this

chamnit, 2017-02-04:
Are you using UGS platform? Have you tried Grbl Panel or BCNC? Any difference?

Russellfam, 2017-02-05:
using UGS not UGS platform and I've not tried the others. I will do so today and let you know

Russellfam, 2017-02-05:
OK so I've been running on the UGS platform and it seems ok on that. being a Sunday I cannot fire up the router and dust extractor I'll check tomorrow and let you know.

Russellfam, 2017-02-05:
I tried with a different cable a short while ago and that seems to have solved the issue. I dont have any with ferrite, but the one I used is a lot shorter.

closed 2017-02-05 by Russellfam

chamnit, 2017-02-05:
@Russellfam : Thanks for the update. I think only UGS platform supports Grbl v1.1 so this was probably the likely cause.

electrokean, 2017-02-05:
@chamnit from everything @Russellfam describes this was an EMI issue.  I've seen it several times on various machines with the USB just dropping communications when the spindle starts running.  Sometimes the serial port disappears, but not always.  Changing the cable, or adding a clip-on ferrite almost always fixes it, but sometimes it takes re-routing the wiring and fixing grounding.
The issues with old version of UGS wouldn't have helped, and may have added to difficulty pinpointing the true issue.  EMI is harder to debug than software :)

chamnit, 2017-02-06:
@electrokean : I suppose, but if it was running perfectly prior to updating to v1.1, then there is a good chance that it had to something to do with that update. AFAIK, regular UGS doesn't support Grbl v1.1 and only UGS platform does. It would likely explain the failure near the beginning of a job, when older UGS got some feedback it wasn't expecting with the new v1.1 interface. Either way, USB EMI is always a suspect, and it seems like it's been solved.