grbl/grbl #1177: Runabout Line/Machine

cacope, 2017-02-06:
I was drawing a design using an Open Builds C-Beam CNC Mill using a CNC XPro control board running GRBL 0.9i. When drawing the outer perimeter of the right side of the crown, the pen did not stop at the proper location but continued and eventually ran into the limit switch, as shown in the photograph below.
[image]

After reviewing the gcode file, that was produced using inkscape and makercam,  the code stopped at the commented out G2 lines of code in the section of code below:
`(Generated by PartKam Version 0.05)

G20 G90 G40

(follow path 1)
G0 Z0.5
(T0 M6)
G17
M3
G0 X9.6728 Y1.6736 
G3 X9.6655 Y1.6138 I2.0421 J-0.2789
G2 X9.6591 Y1.5595 I-2.4615 J0.2631
G2 X9.655 Y1.5326 I-0.6064 J0.0784
G2 X9.6233 Y1.3867 I-1.7961 J0.3138
G2 X9.5169 Y1.0026 I-12.5677 J3.275
(G2 X9.5151 Y1.003 I575.8389 J3144.4886)
(G2 X9.5115 Y1.0036 I2301.836 J12569.6144)
G3 X9.014 Y1.0934 I-17.817 J-97.2928
G3 X8.756 Y1.138 I-4.6455 J-26.1203
G3 X8.325 Y1.2062 I-3.9146 J-23.3448
G3 X7.9432 Y1.2589 I-2.7188 J-18.2727
G3 X6.985 Y1.3549 I-2.2534 J-17.6704
G3 X5.9175 Y1.4066 I-1.5955 J-21.8844
G0 Z0.5
(Left Half Crown Outline)
G0 X5.9182 Y8.5324
G1 Z-0.0098 F320
G3 X5.7843 Y8.6498 I-0.6466 J-0.602 F320
G0 Z0.5
M5
M30
`
What is wrong with these lines of Code?

When commented out, the code ran normally.

LETARTARE, 2017-02-06:
M3 ->  M3 F10

chamnit, 2017-02-06:
@cacope : Try using the most current v0.9 or the new v1.1. I recall there was a bug fix related to really horrible arcs, like the ones you commented. These have a radius of more than 100 to 300 meters and a travel distance of about 20 microns. I can't say if it'll fix your issue, because arcs, as defined by the gcode standard, are horribly posed. They are inherent to numerical error. Some CAM programs can output just G1 commands and no G2/3 arcs. Try that if you still have issues.

closed 2017-02-06 by chamnit

cacope, 2017-02-07:
Thank You, I also reviewed my inkscape file and found the start of the overrun was at a node with long handle and minor angle. I straighten the node and shortened the handle and re-calculated g-code in MakerCam. Then, I re-ran the ink drawing on the cnc and found it ran properly with no runaway.