grbl/grbl #1179: Mirror X axis in corexy

trbLeeciN, 2017-02-10:
Hello, first of all thanks for grbl (: you are a life saver.
My question is, I am using corexy with grbl 1.1,When $3=0, my X axis becomes Y axis, and X axis becomes Y axis but in correct directions. When I invert direction($3=2), Y axis moves correct but X axis goes inverse, I need to mirror X axis, all my gcodes becomes mirrored. Also when I try $3=1, X axis moves correct but Y axis goes inverse.  I hope I made myself clear, it is a little bit complicated ):

trbLeeciN, 2017-02-10:
Well, I tried to switch motors, and it worked, Who needs to mirror X or flip Y axis, try to switch motors pin.

closed 2017-02-10 by trbLeeciN

jahnj0584, 2017-02-10:
You'll want to apply an invert mask. The code is on the wiki for confirming
grbl. It's more then just 1 and 0

chamnit, 2017-02-10:
@trbLeeciN : CoreXY setup is very non-forgiving. The motor location and rotations have to be exactly as described on their theory page. Otherwise, the machine will move in the wrong directions. Inverting the direction pins is often the first thing to try. Good to hear it worked out.