grbl/grbl #1184: Visualization doesn't work - Win10 Intel(R) Q45/Q43 Express Chipset (MS Corp - WDDM 1.1)

rkreiple, 2017-02-20:
2 PC's, no visualization.

3rd same PC with NVidia GT610 works.

Apparently the Intel(R) Q45/Q43 Express Chipset (Microsoft Corporation - WDDM 1.1) driver won't handle it, Driver is the latest.

gerritv, 2017-02-20:
This has nothing to do with Grbl itself, it is a GUI problem. You should post your issue in the relevant GUI Github

closed 2017-02-20 by chamnit

rkreiple, 2017-02-20:
Can you point me to the 'relevant GUI Github'?

gerritv, 2017-02-20:
You haven't provided any information to help you. Look at the name of the program you are using to 'visualize' and search for its web page. It is certainly not Grbl.

rkreiple, 2017-02-20:
My apologies if I am in the wrong place, I thought this is the correct place.

I am using grblControl0.8, this is the screen shot of it running on system with the display problem.

[image]

gerritv, 2017-02-20:
You can find that  on github.com.