grbl/grbl #1187: SpnEn Not working on CNC Shield V3.00 whith gbrl 1.1f

jhosorios, 2017-02-25:
Hello, My Name is Jorge. I've Just try to reconfigure a CNC (shield V3.00) to a better performance and updated the firmware from Grbl0.8 to 1.1f. Everything it´s OK, the CNC works perfectly on the table, but now I have an issue (using Candle to send gcodes):
Whit the grbl0.8c and Universalgcodesender the SpnEn pins used to worked ok to turn on/off the spindle using a relay. But whit the new software those pins just stay on High. How can I automate this again?

jhosorios, 2017-02-25:
I found a solution:
Use the z+/gnd pins as on/off, on the command window set  "S1000"  and "eureka", The spindle turns on and off by itself.

chamnit, 2017-02-25:
Pins 11 and 12 swapped in v0.9 and after. This is to access the PWM hardware on pin 11 for spindle speed and zlimit is now pin 12. See the pin diagram in the wiki.

closed 2017-02-25 by chamnit