grbl/grbl #1188: Homing direction

vyuke, 2017-02-25:
hello,I use the grbl version number is 0.9j, the GUI software is grblControl 0.8.4, i encountered a problem and i can not fix it in the homing function when the axis motion is wrong and i am modifying the direction of the code but The direction of movement does not change, please help me
[image]

electrokean, 2017-02-25:
Don't change the default in the source code, change the configuration parameter $3
See https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

electrokean, 2017-02-25:
Sorry, I mean $23

chamnit, 2017-02-25:
You need to change settings with $ commands. Not the defaults file. This sets the EEPROM upon first flashing or a forced EEPROM reset. As usual, please try reading the wiki. This all covered there.

closed 2017-02-25 by chamnit