grbl/grbl #1190: gnea/grbl-Mega multiple errors compiling

randyf1965, 2017-03-01:
I am trying to upload grbl-Mega to my Mega2560. I have downloaded the zip file, extracted the zip and added the grbl folder into the Arduino library. When I try to compile I get the following:

Arduino: 1.5.7 (Windows 8), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

grbl\coolant_control.c.o: In function `coolant_stop':
C:\Users\Randy\Documents\Arduino\libraries\grbl/coolant_control.c:39: multiple definition of `coolant_get_state'
coolant_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/coolant_control.c:39: first defined here
/Arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
grbl\coolant_control.c.o: In function `coolant_sync':
C:\Users\Randy\Documents\Arduino\libraries\grbl/coolant_control.c:61: multiple definition of `coolant_stop'
coolant_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/coolant_control.c:61: first defined here
grbl\coolant_control.c.o: In function `coolant_init':
C:\Users\Randy\Documents\Arduino\libraries\grbl/coolant_control.c:26: multiple definition of `coolant_init'
coolant_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/coolant_control.c:26: first defined here
grbl\coolant_control.c.o: In function `coolant_set_state':
C:\Users\Randy\Documents\Arduino\libraries\grbl/coolant_control.c:77: multiple definition of `coolant_set_state'
coolant_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/coolant_control.c:77: first defined here
grbl\coolant_control.c.o: In function `coolant_sync':
C:\Users\Randy\Documents\Arduino\libraries\grbl/coolant_control.c:109: multiple definition of `coolant_sync'
coolant_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/coolant_control.c:109: first defined here
grbl\eeprom.c.o: In function `eeprom_get_char':
C:\Users\Randy\Documents\Arduino\libraries\grbl/eeprom.c:51: multiple definition of `eeprom_get_char'
eeprom.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/eeprom.c:51: first defined here
grbl\eeprom.c.o: In function `eeprom_put_char':
C:\Users\Randy\Documents\Arduino\libraries\grbl/eeprom.c:79: multiple definition of `eeprom_put_char'
eeprom.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/eeprom.c:79: first defined here
grbl\eeprom.c.o: In function `memcpy_to_eeprom_with_checksum':
C:\Users\Randy\Documents\Arduino\libraries\grbl/eeprom.c:130: multiple definition of `memcpy_to_eeprom_with_checksum'
eeprom.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/eeprom.c:130: first defined here
grbl\eeprom.c.o: In function `memcpy_from_eeprom_with_checksum':
C:\Users\Randy\Documents\Arduino\libraries\grbl/eeprom.c:140: multiple definition of `memcpy_from_eeprom_with_checksum'
eeprom.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/eeprom.c:140: first defined here
grbl\gcode.c.o: In function `gc_sync_position':
C:\Users\Randy\Documents\Arduino\libraries\grbl/gcode.c:44: multiple definition of `gc_init'
gcode.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/gcode.c:44: first defined here
grbl\gcode.c.o: In function `gc_execute_line':
C:\Users\Randy\Documents\Arduino\libraries\grbl/gcode.c:57: multiple definition of `gc_sync_position'
gcode.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/gcode.c:57: first defined here
grbl\gcode.c.o: In function `gc_execute_line':
C:\Users\Randy\Documents\Arduino\libraries\grbl/gcode.c:67: multiple definition of `gc_execute_line'
gcode.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/gcode.c:67: first defined here
grbl\jog.c.o: In function `jog_execute':
C:\Users\Randy\Documents\Arduino\libraries\grbl/jog.c:26: multiple definition of `jog_execute'
jog.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/jog.c:26: first defined here
grbl\limits.c.o: In function `limits_disable':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:61: multiple definition of `limits_disable'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:61: first defined here
grbl\limits.c.o: In function `limits_init':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:35: multiple definition of `limits_init'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:35: first defined here
grbl\limits.c.o: In function `__vector_9':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:70: multiple definition of `limits_get_state'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:70: first defined here
grbl\limits.c.o: In function `limits_go_home':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:100: multiple definition of `__vector_9'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:100: first defined here
grbl\limits.c.o: In function `limits_go_home':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:148: multiple definition of `limits_go_home'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:148: first defined here
grbl\limits.c.o: In function `limits_soft_check':
C:\Users\Randy\Documents\Arduino\libraries\grbl/limits.c:342: multiple definition of `limits_soft_check'
limits.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/limits.c:342: first defined here
grbl\main.c.o: In function `main':
C:\Users\Randy\Documents\Arduino\libraries\grbl/main.c:32: multiple definition of `main'
main.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/main.c:32: first defined here
grbl\motion_control.c.o: In function `mc_homing_cycle':
C:\Users\Randy\Documents\Arduino\libraries\grbl/motion_control.c:33: multiple definition of `mc_line'
motion_control.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/motion_control.c:33: first defined here
grbl\motion_control.c.o: In function `mc_probe_cycle':
C:\Users\Randy\Documents\Arduino\libraries\grbl/motion_control.c:82: multiple definition of `mc_arc'
[... 356 lines cut ...]
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:353: multiple definition of `system_clear_exec_state_flag'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:353: first defined here
grbl\system.c.o: In function `system_set_exec_alarm':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:360: multiple definition of `system_set_exec_alarm'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:360: first defined here
grbl\system.c.o: In function `system_clear_exec_alarm':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:367: multiple definition of `system_clear_exec_alarm'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:367: first defined here
grbl\system.c.o: In function `system_set_exec_motion_override_flag':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:374: multiple definition of `system_set_exec_motion_override_flag'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:374: first defined here
grbl\system.c.o: In function `system_set_exec_accessory_override_flag':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:381: multiple definition of `system_set_exec_accessory_override_flag'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:381: first defined here
grbl\system.c.o: In function `system_clear_exec_motion_overrides':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:388: multiple definition of `system_clear_exec_motion_overrides'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:388: first defined here
grbl\system.c.o: In function `system_clear_exec_accessory_overrides':
C:\Users\Randy\Documents\Arduino\libraries\grbl/system.c:395: multiple definition of `system_clear_exec_accessory_overrides'
system.c.o:C:\Users\Randy\AppData\Local\Temp\build540589733081590121.tmp/system.c:395: first defined here
collect2: error: ld returned 1 exit status

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

chamnit, 2017-03-01:
This usually means you have not removed prior Grbl versions in theArduino library path. Meaning you have more than one.

closed 2017-03-01 by chamnit

randyf1965, 2017-03-01:
Thanks! I was able to compile by going into Files-Examples-grbl-grblUpload