grbl/grbl #1192: hello friends, iwould like to use brushed dc motor with encoder feedback insted of stepper how to modify gbrl to do it?

farhanalbadran, 2017-03-02:


Norwegian-Gadgetman, 2017-03-02:
The short of it is, you don't. 

The long story is that it would require a complete rewrite. 
If you wish to even try, take a look at this page:
https://www.youmagine.com/designs/dc-motor-closed-loop-control-software

farhanalbadran, 2017-03-02:
thank you for your reply,the reason why I need to modify gbrl is that it is g code compatible else there is a very simple Arduino cnc code that uses dc motors with feedback , http://homofaciens.de/technics-machines-cnc-v2_en.htm

farhanalbadran, 2017-03-02:
so what i am asking is that where is the part that I can modify  in GBRL source code to replace stepper motors with dc motors that is going to require pwm pin as well as direction pin and two pins for encoder