grbl/grbl #1193: Allow other boards with other pin assignments

yellobello, 2017-03-04:
Hi,

coming from the printing world with marlin,I am quite new to grbl. 
I have purchased a china board and the first thing I did was upgrade it to the latest grbl version,
only to find out that the guys used a different pin mapping than grbl does for the official shield.
Since I have got it working with grbl now, I thought why not ask you guys if it can be allowed to grbl master as an option for a different board just like marlin does for many boards. 
But the issue might be that the board is not open source hardware I guess.
btw, it´s the "Laseraxe" board I´m talking about.

jahnj0584, 2017-03-04:
You mean being able to change the pinout, like many of the mach3 breakout
boards do? I think you can simply recompile grbl with the changes made in
the text file and it'll do the same thing.

yellobello, 2017-03-04:
Yes, thats what I did, but I thought it would be easier for people like me that run into the same problem to keep multiple cpumap files for different boards each and then select the desired file in a define before compile. ..just a thought

109JB, 2017-03-04:
You can already edit/add a pin mapping to cpu_map.h and then change the following line in config.h to use your desired pin mapping.:
#define CPU_MAP_ATMEGA328P

yellobello, 2017-03-08:
Yes, thats what I did.. I just thought it would be good to implement some boards into 
the main grbl codebase.

Other thing, kinda related:

Are the pin numbers somehow bit-shifted around? I don´t get the way how some pins are assigned.. for example the pwm'd spindle pin, which should be on arduino 11, right?
let´s see:

from cpu_map.h:
`// Define spindle enable and spindle direction output pins.
  #define SPINDLE_ENABLE_DDR    DDRB
  #define SPINDLE_ENABLE_PORT   PORTB
  // Z Limit pin and spindle PWM/enable pin swapped to access hardware PWM on Pin 11.
  #ifdef VARIABLE_SPINDLE
    #ifdef USE_SPINDLE_DIR_AS_ENABLE_PIN
      // If enabled, spindle direction pin now used as spindle enable, while PWM remains on D11.
      #define SPINDLE_ENABLE_BIT    5  // Uno Digital Pin 13 (NOTE: D13 can't be pulled-high input due to LED.)
    #else
      #define SPINDLE_ENABLE_BIT    3  // Uno Digital Pin 11
    #endif`

pins 3 and 5 have been assigned to Y_STEP_BIT and Z_STEP_BIT some lines above.
ok, so I have a look whats going on with SPINDLE_ENABLE_BIT, SPINDLE_ENABLE_DDR  and  DDRB in spindle_control.c  but then I completely loose track because it´s all bit-shifting around..

I don´t get it.. sorry

langwadt, 2017-03-08:
@yellobello same bit number different port

duronebis, 2017-03-09:
I see a big problem if you want to use a shield with the step and dir pin located on different Ports (like ramps, for example). If I don't go wrong Grbl wants all the pin of the same type on the same port.
I used the Marlin way (fastio.h) to solve this problem, but this require deeper changes in the source code and probably little longer time to execute the stepper interrupt.
As I also added the 4th and 5th axes, I limited the max frequency to about 20kHz, for my foam cutting project this is enough.
I'll then try to move to Arduino Due, to try higher frequency for larger foam cutting machines.

electrokean, 2017-03-09:
@yellobello the grbl pinout has been optimised to get the highest practical performance, and as @duronebis says changing that is not trivial.  And for someone like @chamnit it is a support nightmare as there is no way he could test and retest all the user-submitted pin mappings.  But being open source, you are completely free to make these changes yourself, and even publish them for others.

Also note that there is no obvious mapping between Arduino pin numbering and port names/bits (or IC pins for that matter).  Check out one of the great pin-out diagrams like http://www.pighixxx.com/test/portfolio-items/uno/ which shows all the gory detail.

chamnit, 2017-03-11:
FYI, the Grbl based project is including multiple pin assignments with a hardware abstraction. But this will not include the 328p Uno. Only Mega, ARM, or any micro with enough ram and flash. Ports will not be supported by myself, but users interested in maintaining a particular port for the community.

closed 2017-03-11 by chamnit