grbl/grbl #1194: XY plane leveling

jmesar, 2017-03-08:
Is Clear the GRBL definition: “high performance g-code-parser and CNC milling controller” but as human we always want more…
I end my first CNC, and found that the XY plane have 0.4mm error, not bad for first try and home tools.  Anyway I found BCNC and Chilipeppr can compensate it.
Do exist a “perfect” plane alignment? I guess not, so the plane alignment should be part of the machine as is the GRBL itself, then would be a good thing that the alignment task were inside the GRGL, performed only once at the commissioning and never again.
Arduino UNO perhaps could make something like this:
X’=Xa11+Ya12+Za13
Y’=Xa21+Ya22+Za23
Z’=Xa31+Ya32+Za33
Where a11, a112…..  Are so many years away in my vector analysis classes…..
Perhaps Arduino UNO can not, but other could, or perhaps a Raspberry Pi

jahnj0584, 2017-03-08:
Can you not just surface your spoilboard and it will be flat and parallel

lalo-uy, 2017-03-08:
The Gcode sender, as bCNC, can do that .
It is called autolevel

jmesar, 2017-03-09:
@jahnj0584 my "spoilboard" is on aluminum, flattering would be a little hard, but is a valid solution.  
Yesterday I saw a more "Clean" solution in https://www.youtube.com/watch?v=DmuipSuDNAs&t=17s with aluminum foil to level.   Anyway, a script on GRBL based on G38.2 command and coordinates transformation would be more accurate. 

@lalo-uy I did not find any autolevel on Gcode Sender.  bCNC has and interesting feature: camera to track and center with OpenCV, but i have not be able to install OpenCV.

109JB, 2017-03-09:
> Anyway, a script on GRBL based on G38.2 command and coordinates transformation would be more accurate.

I would totally disagree with this statement. It is always better to have a straight machine rather than artificial compensation.

jmesar, 2017-03-11:
As everything it depends of…. What are going to do with your CNC.   I build mine CNC to wood carving and second to try engraving, for wood carving the plane misalignment is easy to solve from Aspire or ArtCam, for engraving the gap is as bigger as work itself.
With 200 steps/rev, 3 or 4 decimal digit in GRBL, and been every coordinate transformation an independent event, the “error” is not accumulative, so you don’t need to go at Lorenz chaos’s findings. 
Otherwise, perfect plane alignment does not exist in nature, it just do in Euclid’s geometry in the Plato cave …If you run a flattering bit over the surface, you still have the MDF irregularities, and also the CNC backlash error.  
Which error would be bigger?  I don’t know.  My spindle is just a Dremel, and in my bits arsenal I still don’t have a flattering bit, so software translation is a “clean” solution.  I would like the developer team think about.

chamnit, 2017-03-11:
Do yourself a favor and align your axes. If you can't, upgrade your machine. It'll save you a lot of time and energy. This is the time tested way to improve machine performance. 

That said, I don't disagree with you about calibrating the workspace in firmware. Professional machines do this as a service to make older machines more accurate after the ways have worn. They use highly accurate calibration bars and measure in multiple directions from multiple points throughout the machine space. The process takes a whole day. This type data collectionand tooling is not available to a typical hobbyist, especially ones that can't afford a properly built machine.

closed 2017-03-11 by chamnit

jmesar, 2017-03-19:
@chamnit So You have the post, you can convert a typical hobbyist machine in a Professional grade machines!!!!