grbl/grbl #1196: How can the go home command ($H) be included in an autostart string ($Nx)?

mlmccauley, 2017-03-09:
The line
$N=$H faults out.

For reference, I am using 0.9j, compiled locally, so making changes to the source is not a problem.

Thanks!

chamnit, 2017-03-09:
It's not allowed. Autostarting any motion via a startup line is a very bad idea. This can lead to crashes.

closed 2017-03-09 by chamnit

109JB, 2017-03-09:
You definitely don't want to do that for safety reasons. The intent of the $N commands is to set default states, and it is inadvisable to include anything that would result in any kind of motion.

Also the homing should only need to be done once after a power on or hard reset. A soft reset doesn't lose position, but the $N commands run on each reset whether a soft rest or a hard reset or a power on.

Also, the $N command has to be a legitimate G-code string and can't include any special grbl commands.

mlmccauley, 2017-03-09:
Understood on only G codes being legal. The software informed me of that in short order.
My application is a 3D printer, not a machine tool. I am only using the Z axis. A stepper on Z drives the build platform from home down as the liquid resin is hardened in layers by UV light, nothing to smash up.

Anyway, thanks for the replies. I'll figure out another way to get to where I need to be.