grbl/grbl #1197: 4 axis?

MohammadMubin, 2017-03-09:
Hi everyone i got my CNC working under GRBL but my 4 axis is laying around and i would to find out when GRBL will have support for 4 axis.
Thank You.

electrokean, 2017-03-09:
This is never going to happen on the Arduino Uno (ATmega328) version of grbl, as there is not enough IO pins.  There are several non-official 4-AXIS or 6-AXIS forks around for the Arduino Mega (ATmega1280/2560), and can be found with a little search on the closed issues here or on Google (including mine).
Future support of 4 or more axis will probable be under one of the new repos at https://github.com/gnea/

109JB, 2017-03-10:
Actually, you could get enough pins for 4 axes just by combining all limits on a single pin.  That would free up 2 pins that could be used for a 4th axis.  Combining the limit pins requires changing the homing sequence, so that each axis is homed individually, but I have done the single pin for all limits thing and it worked fine. The details can be seen in the grbl v1.1 github in this post https://github.com/gnea/grbl/issues/37

electrokean, 2017-03-10:
@109JB yeah, that is kind of true - the limit pins are on Port B though, and the step and direction pins are on Port D.  The way things are structured you really want all step pins & direction pins on the same ports - so it isn't a trivial change, and may limit use of things like the PWM output.  The other 4/6-AXIS forks use a Mega with separates 8-bit ports for step and direction, potentially allowing 8 axes.  Each additional axis will limit overall speed of the controller though.

MohammadMubin, 2017-03-10:
i do not mind spindle not being there which frees up 2 pins which is enough for Step and Dir pins. But the modded ones which i saw have no proper software to control with. 
(P.S that was very fast reply)

electrokean, 2017-03-10:
My 6-AXIS fork for the Mega is [here](https://github.com/electrokean/grbl/tree/6-AXIS) - it is usable, but it is reasonably out of date now.
Another one is [here](https://github.com/EliteEng/grbl/tree/6-AXIS), also probably quite out of date.
I wouldn't recommend trying to get 4+ axis out of a Uno/Nano unless you don't have the option to switch to a Mega for some reason.

I don't know of any Grbl GUIs that support 4 or 6 axis, but there could be one.  There are several open source GUIs so you can always fork one and customise one.  I actually don't use a standard GUI, but instead send G-code from a custom Windows application, or in one machine from another Arduino Mega.

CAM software is another matter completely.  CAM for true 4+ axis machining can be quite expensive.  I just use my multi-axis version for customised industrial robots, not machining.  And most people I know using a 4th axis, just use it for a rotating axis in place of X or Y for engraving on cylinders - so they don't really need 4-axis control at all.

MohammadMubin, 2017-03-10:
Alright Thank you i will check them too see.

duronebis, 2017-03-10:
MohammadMubin : I'm developing Cad/Cam/Cnc 4 axes foam cutting applications for a company, the Cnc includes also support for Arduino, but  they are not open source. If you like You can contact me to info@devcad.com, as I think giving more info here is not fair.

109JB, 2017-03-10:
@electrokean   I am not up on github tools. Is there a way to see a diff of your 6 axis fork and the version you forked from?

electrokean, 2017-03-11:
@109JB yep, there is a compare button up there on the right if you visit my fork.
It will take you to https://github.com/grbl/grbl/compare/master...electrokean:6-AXIS
And you can select from drop downs which forks/trees to compare
Personally, I prefer to download snapshots and use a local visual diff program like ExamDiff or BeyondCompare

MohammadMubin, 2017-03-11:
So is there any CAM Software's which are free,supports 4 axis and  open source?

MohammadMubin, 2017-03-11:
Dang it! well i will stick to my 3 axis router.

closed 2017-03-11 by chamnit

X3msnake, 2017-05-30:
@electrokean 

FYI - Fusion 360 is free for makers and companies up to 100K anual and it is the full 5 axis version

electrokean, 2017-05-30:
@X3msnake Nice.  Of course 100k annual is reasonably a small company...

X3msnake, 2017-05-30:
Sure, but if you make more than that you surely can afford the 125.eur
anual fee for the 3 axis version or the 1.250.eur for the full :)

langwadt, 2017-05-30:
if you are above 100k you can probably afford the 320€/year ;)

X3msnake, 2017-05-30:
Here's the full table of with free and paid tiers as it is today

The following table describes the benefits for various subscription types:

  *Trial* *Startup* *Student* *Fusion 360 Standard* *Fusion 360 Early
Adopter* *Fusion 360 Ultimate*
*Price* Free Free Free $300 annual OR $40 month-to-month $300 annual OR $40
month-to-month $1500 annual OR $190 month-to-month
*Eligibility Criteria* For 30-day free access for evaluation For Hobbyists
and Startups making less than $100K per year For Students and Educators For
individuals and teams making greater than $100K per year For individuals
and teams making greater than $100K per year For individuals and teams
making greater than $100K per year
* Software Benefits* Fusion 360 ULT Fusion 360 ULT Fusion 360 ULT Fusion
360 Std Fusion 360 ULT Fusion 360 ULT
*Cloud Credits** 25CC
Valid during trial ** ** One-time grant of 100CC on first purchase  One-time
grant of 100CC on first purchase AND 1000CC on new purchase and on
renewal One-time
grant of 100CC on first purchase AND 1000CC on new purchase and on renewal*
CC for annual contracts
** Autodesk is pleased to support free educational/hobby/startup usage of
our software and cloud services and encourages customers with an Education
or Startup entitlement to practice responsible consumption of cloud
services to help ensure good service level provision across its user base.
Customers with an Education or Startup entitlement who regularly
demonstrate excessive cloud credit consumption beyond market average may
face restrictions. For more information about free entitlements, visit
theEducation Special Terms and the Autodesk Web Services Terms of Service.

https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Autodesk-Fusion-360-Changes-to-Benefits-November-2016.html

devekrehbiel, 2018-06-10:
Have we come any further on this issue in the last year and a half? I am building a robotic arm and need 5 axis. I am not a programmer, so does anyone have now, or is working on, a MORE axis solution than just three? I do not know why, but I tried the Mega 2560 version of GRBL and couldnt get the servos to move. I am back to the Uno until I can figure out why. Thanks everyone. We NEED more Axis! My hope is GRBL and UGS can make this happen soon.

109JB, 2018-06-10:
https://github.com/LETARTARE/grblQ-Mega-4axes
https://github.com/dguerizec/grbl-Mega-4axis/tree/0046fa9d386b0c5fa219a0cede47b5c339ea9cc8

MohammadMubin, 2018-06-11:
i Use fusion 360 for all my Routes but the fourth axis in Fusion is weird and the gcode which it spits out is not in the right form too but Fusion 360 works great with 3 axis and great simulation :D

MohammadMubin, 2018-06-11:
@109JB  that's great! illl have a look at them 

@devekrehbiel  #MoreAXIS petition is coming right up (jk grbl is great)

cri-s, 2018-06-11:
There exist 6 Axis version too.

devekrehbiel, 2018-06-11:
I am very new to all this and find the instructions very hard to follow. After you use xloader for installing the hex file on the mega, what do you use to control all four or six axis? UGS only supports three.

devekrehbiel, 2018-06-12:
Thanks for this information. I am still not sure how to make GRBL work alone. I realize commands can be sent via the serial monitor on Arduino. I just do not know how to send text files of commands that repeat to make a robotic arm work consistently. BCNC is something I have no idea how to make a python effort work in windows, and the instructions just assume you wrote the dang thing. Its just a 3 axis solution too. There is no winning!

I am not trying to be mean.. I am a nice guy and very appreciative of everyones efforts. Maybe I am getting old, but there are 17 "axis" in just the human hand. Nevermind the entire body. If a person were to go any further than a single finger, we are all currently SOL. We badly need the programmers of these efforts to think in terms of adding MANY axis. How many axis do you need? And then it repeats the same code labeled differently to make this work. Universal Gcode Sender is a great option, but for some reason, we are stuck on 3 axis and we cant seem to get past it. I am not a programmer and I do not have a clue on how hard this all is, and I am sure it is HARD. But please, PLEASE think of 3 axis as a good START. As with everything I do for free for the public, my hope is you all can understand that we are dead in the water until we can get MORE AXIS!! Its a work in progress, but devestechnet.com/NewTech is my effort to give back. The robotic arm I have created is not there yet, because I am screwd until we can make more happen. Sorry for the soapbox, but we are in 2018! I was born in 1958. I do not have all day! Thanks again everyone!

chamnit, 2018-06-12:
@devekrehbiel : Grbl has always been a 3-axis project solely to keep things simple as more advanced features like realtime overrides were installed and to fit into the Arduino Uno's processor. There are lots of prior posts about 4+ axes that all say Grbl is moving on to 4+ axes. This coincides with moving to a much larger and faster ARM chip, which has been planned and worked on for years.

devekrehbiel, 2018-06-12:
I can appreciate it takes time to do this, but the Uno is OLD and the Mega2560 could easily handle more axis. I have NO idea what I am talking about, other than I have a robotic arm right now that needs movement on 5 axis and possibly more later on. So I have to wait until we can get the software part of it up to speed. Thanks for the reply, but please think about where this is all going. When a person wants to build a robotic human body, 100's will be needed. Then we will go to a more powerful processor. or. thinking about it, can we integrate multiple Mega2560s to get the job done? We cant be out of ideas at 3 axis and an Uno. Keep up the good work. I am NOT being critical, rather trying to spur a conversation on how to make leaps over this quagmire we seem to be stuck in.

chamnit, 2018-06-12:
It only seems like a quagmire because I haven’t publicly published everything I’ve been doing. There are many good reasons for this that I won’t get into. 

The Arduino Uno is old but it’s prolific. It’s a very low bar for anyone to try CNC. 

The Mega2560 is literally the same speed as the Uno. It only has more RAM, flash, and IO. The processors I’ve been looking at are at least 10x faster or more.

109JB, 2018-06-12:
@devekrehbiel You should realize that @chamnit has a full time job and has evaluate will benefit the most users when he is applying fixes and enhancements to Grbl.  Most of the users are utilizing Grbl on 3 axis cartesian coordinate machines and this is what Grbl was originally written for by the originator. Extending it to more than 3 axes on an Arduino UNO, or equivalent is just not possible due to lack of pins and lack of program space. The Mega does have more pins, and more space, and as noted has been extended to more than 3 axes. Extending beyond this is possible, but I would guess it results in slower performance for every axis added due to the planner needing to account for each additional axis when planning moves. As @chamnit  said, the Mega processor is no faster than the UNO.  @chamnit does a great job and he will get it there, but as said he has a real job and this is volunteer time he spends on Grbl.

As for that 6 axis fork I posted earlier, disregard that one. I have done a bit of testing over the last day and it is really really buggy. So much so that I would not trust it at all.

The 4 axis forks I posted I have done a lot of bench testing and they both seem to work fine.  I haven't had a chance to test either on a real application, but I am confident they would both work.

Your application for a 5 axis robotic arm is not what Grbl was intended for. Doesn't mean it couldn't work for it, but you just need to realize your situation is not the norm and is not in line with most Grbl users. It is certainly not intended for human body simulation as you also mentioned. That would seem to need much more computing power than a typical MCU. I would also think that with the proliferation of hobby kits for robotic arms that there must be some firmware/software available for running these. If not, then you may need to look to other solutions such as PC based software like LinuxCNC, which can do 9 axes with the right equipment.

devekrehbiel, 2018-06-12:
I understand and I am not trying to offend anyone, but I am trying to explore all of the avenues for 5 axis and there really are none. Since the concept of Axis' really means controlling stepper motors, I vote for the goal to be choosing the number of steppers you want to control between 0 and the max capability of the processor being used. This concept was born on the idea of Axis, XYZ, but we are way beyond that now. In any case, I will drop whatever I am doing and help out testing for anyone working on a solution. We all have jobs and lives, but this is long overdue, so let me know how I can help. THANKS to all who have gotten us this far.

Other than programming the text into Serial.println, how do you send a long string of commands to the motors using just GRBL with no GUI? I have LOTS to learn.

chamnit, 2018-06-12:
@devekrehbiel : You obviously have a lot to learn about motion control and programming. I suggest not providing strong opinions on things that you are not well versed in. It's a waste of time for everyone.

devekrehbiel, 2018-06-12:
But I learn real fast when others share what they know. Simple questions on how to use the software shouldn't be viewed as a waste of time. Rather than taking offense to what I am suggesting, since it is very valid in the bigger picture, how about we all just get along? I've read every thread everywhere concerning GRBL and there isn't much on exactly how to use it. I can help with that since documentation is what I do, but I sort of need to understand the basics. Once I do, you will be very appreciative of the outcome which I will gladly share when its done.

chamnit, 2018-06-12:
@devekrehbiel : We're not the ones constantly arguing and insisting that we're not trying to offend anyone. Then going on, being abrasive, and demanding things that have already been answered in prior posts. I'm sorry that the next version is not out now. You will have to wait like everyone else to get it. Or, move on to another CNC/robotics project that is going to do what you want or need.

As far as a waste of time point and speaking for myself only, I'm not here to hold your hand through everything. Even the most basic of things that you can search for online and get the answers to easily. It takes time to write a message, go through the process of finding out the problem, and go back and forth until the problem is solved. This isn't a customer service line. Yes, I have a day job. Yes, I have a family. Yes, I work on Grbl in my spare time. Pointless online arguments takes time away from the latter.

109JB, 2018-06-12:
> But I learn real fast when others share what they know. Simple questions on how to use the software shouldn't be viewed as a waste of time. 

https://github.com/grbl/grbl/wiki
https://github.com/grbl/grbl/wiki/Compiling-Grbl
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

and the list goes on.

> I've read every thread everywhere concerning GRBL and there isn't much on exactly how to use it.

Grbl is a firmware that takes a stream of text via the Arduino serial port. The text stream has to conform to the allowable Grbl commands and G-code standards which are documented in the WIKI.  I fail to see what is so hard to understand about that.  How that communication happens is here:

https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

Actually it seems pretty well documented to me.  I have been able to write my own GUI's for Grbl by referring to these WIKI pages.

> Since the concept of Axis' really means controlling stepper motors, I vote for the goal to be choosing the number of steppers you want to control between 0 and the max capability of the processor being used. 

Actually that is not true. As Grbl is written it is predicated on 3 orthogonal linear axes. That is how its kinematic engine is written. LinuxCNC as an example has a kinematic engine that can be modified to account for non-orthogonal axes (such as a robotic arm). As another example, Grbl would not be good at something like a Delta printer without an interface that takes care of the kinematics. A delta printer still has only 3 motors for motion, but all 3 motors turn in unison because of the kinematics of the delta design. Same thing for a polar type printer. It is also the same for a core XY type machine except that someone that was interested in coreXY modified Grbl's kinematic engine to account for the kinematics of the coreXY layout and did a good enough job that it found its way into the main distribution. So, saying that the number of axes is just related to the number of motors is not correct.

Incorporating additional axes is not as trivial as it sounds. Is the new axis rotational or linear? does it need to work independently or does it need to move synchronously with other axes? How is coordinated motion handled? Which axes determine motion planning in terms of feed rates etc?  These all need to be dealt with when adding an axis.  The 4th axis forks I linked work, but they don't work like other 4th axis CNC machines. Is it wrong? Not necessarily, but it all needs consideration. So adding axes is not just a simple matter of doing it either.

As for the number of axes a processor is capable of, it has already been said that for the UNO that number is 3. Yes, now there are many other MCU's out there that would be capable of more, but Grbl's development started on the 328p. It isn't just a matter of uploading it to another MCU, the code has to be written for the MCU it is to be uploaded to. That is why Sonny is working to re-vamp Grbl with a hardware abstraction layer, so porting to a different MCU isn't such an arduous task as it is now.  But re-vamping Grbl with the hal and adding the features IS an arduous task and I for one am thankful that anyone is even willing to do it in the first place. As for further development, the ARM version, which will include more axes and features and be easier to customize is in the works but isn't available yet. That is just a fact of life that has to be dealt with.  I completely understand Sonny's reasoning for this because premature releasing it would just lead to a nightmare of posts with "why didn't you do this?" or "where is that?", etc.  It needs to have a least a certain amount of testing and debugging before any kind of public release.

Believe it or not we are trying to help you but your views on how simple things are is just not in touch with reality.