grbl/grbl #1198: can not get homing to work with grbl servo

nikki-reprap, 2017-03-13:
Hi I am having some problems configuring grbl endstops i have them setup as NO I have disabled hard limits and enabled homing and soft limits and edited the config file for corexy homing but it will not home it moves the x carriage to the endstop then moves the y axis about 10mm but then the motors just start humming.This is the version i am using. https://github.com/robottini/grbl-servo

Thanks Nikki

nikki-reprap, 2017-03-13:
I have managed to fix it by copying spindle_control.c and spindle_control.h to v0.9j and then editing the config.h file. the only problem I have now is when i send gcode after homing the machine moves to the opposite end of the axis and hits the end.

Faisalbelwadi, 2017-06-13:
Did you find any solutions for your problem?
I am having the EXACT same problem (homing) that you are facing with my 4xiDraw( a drawing machine 2axis) . 
Can you send your grbl folder with the modifications  you've done to make it work and and other necessary details

chamnit, 2017-06-13:
Not a supported issue with mainstream Grbl. Closing.

closed 2017-06-13 by chamnit