grbl/grbl #1202: Reversal of Axis Direction

QPatterson, 2017-03-16:
When trying to operate my cnc the X-Axis will switch directions no matter what direction the command is given in (i.e. if it is told to step 10 mm in the minus directions 5 times after the 3rd it will switch and go in the positive), but in the universal g-code sender the output says it is still continuing in the minus direction. It has no correlation of reaching a zero or a boundary and appears to do so at random. The Y and Z axis do not have this problem.

electrokean, 2017-03-16:
Either you have a problem with the wiring of the direction signal from the Arduino to the stepper driver, or you have a faulty direction input on the stepper driver.  I guess it could also be a faulty Arduino/ATmega output pin - but the other two are easier to test/eliminate.

First thing to do is follow the wiring, and measure continuity with a multimeter if you have one (with the power off).

Second, you can try swapping stepper drivers if they are all the same (again with power off), and see if the problem stays with the X axis, or moves with the driver.

woale, 2017-03-23:
@QPatterson what setup do you have? 
I experience the same issue, where direction seems random... I'm on Chilipeppr over Raspberry Pi 3 with a CNC Xpro 3 (from Spark Concepts), still in "build phase". Not sure if it's related, but I only noticed this behaviour since adding limit switches.

woale, 2017-03-23:
Solved for me... it was just a loose wire on my motor.

QPatterson, 2017-03-23:
@woale I'm using an old Sherline with NEMA 17s, TB6560 drivers and an arduino uno. I was able to fix the problem it was just a loose wire that had been ripped out.