grbl/grbl #1203: Steps per revolution are off by 2^n

thomasnordquist, 2017-03-20:
I have the issue that my steps/mm are off a lot.
After some testing and a lot of measuring, I realized that compensating with a factor seems to solve the issue.

BUT: It's a very bad solution !

```
  #define DEFAULT_X_STEPS_PER_MM 200.0*2
  #define DEFAULT_Y_STEPS_PER_MM 200.0*4
  #define DEFAULT_Z_STEPS_PER_MM 200.0*16
```

Used hardware:
 - 3 Steppers 17HS16-2004S1 - 1.8° per Step
 - Arduino CNC Shield V3
 -  Proxxon MF70

jahnj0584, 2017-03-20:
Or maybe calibrate it properly....

https://youtu.be/LRagjLewSJg

langwadt, 2017-03-20:
@thomasnordquist check the microstepping jumpers ...

109JB, 2017-03-21:
Why are you changing the code when there is an EEPROM setting for steps/mm for each axis.

thomasnordquist, 2017-03-21:
@jahnj0584 The existing-coordinate table made this step unneccessary.
@109JB I just learned that these settings are persistent.
@langwadt You were 100% right. I set it now for 4 microsteps => 800 steps/mm
one was actually set to 16 microsteps, which resulted in some skipped steps

Afterwards I discovered I still had missing steps and the some steppers sometime gave out completely.

After applying an airflow to the stepper controllers everything seems to be fine !
Thanks to all !

closed 2017-03-21 by thomasnordquist