grbl/grbl #1205: No luck getting CNC to move

CantSayIHave, 2017-03-29:
Hi, I bought one of those DIY CNC kits from China. The most the seller could give me was a bunch of drivers in chinese.

The control board (a Nano copy and 3 A4988 drivers) was initially loaded with GRBL, and though serial port communication was perfect, I could not make it move. I could interface with grbl, change settings, and tell it to move (to which it would reply ok), but no movement.

I since tried uploading grbl via Arduino, no movement. I just now tried flashing the latest hex file through XLoader. As before, GRBL comms work fine, but the motors do nothing.

The board is a 328p. I can't figure out if the fault comes from the hex file I'm uploading, the use of A4988 motor drivers, or incorrect pin mappings due to using a nano. The board looks professional, [here](http://i.imgur.com/co6pQcI.jpg) is a picture of it. I'm at my wit's end on this issue.

109JB, 2017-03-29:
The nano pin mapping is the same as the uno.  It is likely not the upload of grbl since you get responses back from it.

You don't mention your wiring at all.

Do you have a power supply for the steppers attached to the connector at the top left of the photo you posted?

Are you sure the coil pairs are wired correct for the steppers?

Have you tried inverting the enable signal in GRBL EPROMs settings?

Do the motors lock when stepper power is applied?

Those answers will help us start diagnosing. A picture of your wiring setup may help too.

lalo-uy, 2017-03-29:
The motors come with the connectors  or you wire the connectors yourself?

CantSayIHave, 2017-03-30:
I understand your skepticism, but I did of course follow the instructions. [This](http://i.imgur.com/wYZVtJH.jpg) is the current wiring, and [here](http://i.imgur.com/qeJbEBW.jpg) is a close-up of the control board.

The motors came with pre-assembled connectors, and the Z-axis motor came with the connector integrated, so it should definitely be correct.

I tried inverting enable with $4=1, no luck. The motors also do not lock, inverted or not. I'm hoping I'm not the 1/1000 who receives a faulty board.

Norwegian-Gadgetman, 2017-03-30:
Verify that the PSU actually delivers 12V or 24 or whatever, and in the correct polarity. 
Verify that the drivers are mounted the right way. 

Are the drivers heating up at all? 
Adjust the small pots to about midway and check if it 'holds' the motors. 

Those drivers need a heatsink.

109JB, 2017-03-30:
I wouldn't assume the motor wiring is correct. I see the board has the coil commectors labeled A-B-A-B, but motors are generally A-A-B-B. Also most other motor driver connections I have seen are A-A-B-B, so if the wiring to the motor is wrong then it won't work.  It is easy enough to test if you have a meter.  When measuring A-A or B-B there will be resistance measured. If measuring A-B it I'll show an open circuit.

CantSayIHave, 2017-03-31:
I tested the power supply, it supplies 24V with correct polarity.

I then tested the wiring and the results were... interesting. [This](http://i.imgur.com/mXQiEdP.png) is a drawing I made to illustrate my findings as I don't know what to make of them. As you can see, when testing resistance from each control pin to each stepper pin, the A-B sets seemed to each be connected to each other and to the 1-3 2-4 combo. I also tried resistance from control pin to control pin - it seems 2A connects to 2B, and 1A to 1B. I think I'll try switching the middle two, just in case the board manufacturer switched letters and numbers, and report back.

CantSayIHave, 2017-03-31:
Ok, well I just resolved the problem... it does make me feel pretty stupid. The machine came with two power supplies, one 24V 5A and one 12V 3A. The manufacturer specifically told me I did not need the latter, as it was for the laser extension. So I stupidly did not try to plug it in. It most likely powers to motor drivers. Anyway, I plugged it in, everything works. Thanks for all the help though.

As a side note, the machine didn't come with limit switches. Do you think I can install some on the open header set?

lalo-uy, 2017-03-31:
Ofcourse, put NO switches wirh 10kohm pullups and 100nF to ground to filter
noise.

CantSayIHave, 2017-03-31:
Alright, will do. Thanks!

closed 2017-03-31 by CantSayIHave