grbl/grbl #1206: documentation

lakaiola1128, 2017-04-12:
i am asking if you can provide some detailed schemes on how it works to understand quickly and better the grbl code. see an example here http://smoothieware.org/howitworks

109JB, 2017-04-13:
I personally would prefer Sonny work on actual improvements rather than exerting a bunch of time to explain how free software works.  Smoothieboards are commercial products. Grbl is not. I would not expect an open source project to have the same level of support or documentation as a commercial venture.

Having said that, grbl is going to work pretty much in the same manner as the smoothieware stuff. Feel free to make something yourself and add it to the WIKI.

jahnj0584, 2017-04-13:
Pretty sure smoothie runs grbl so it's already explained what you asked

<https://github.com/grbl/grbl/issues/1206#issuecomment-293747192>, or mute
the thread

109JB, 2017-04-13:
Smoothie firmware started from grbl, but it has moved on from there. Smoothie runs on 32-bit ARM boards and has basically been re-written to where it is no longer grbl, but may have pieces of grbl remaining.

lakaiola1128, 2017-04-13:
perhaps it takes time for new users of grbl to write documentation in a forme of schematic diagrame, but for those wh developped the grbl and are working for years on this project, i think, they can write it easely.

lakaiola1128, 2017-04-13:
_"I personally would prefer Sonny work on actual improvements rather than exerting a bunch of time to explain how free software works"_
explaining how it work leads to better improvement, other developers will understand it better and can make contribute to the improvement

darkfibre-nl, 2017-04-13:
@lakaiola : Looking at the current state of the project we can safely
assume Sonny spends his time wisely. If you want more documentation, feel
free to add it @lakaiola. When it's there, perhaps ask if Sonny has time to
review it, and suggest improvements.

mayhem2408, 2017-04-13:
@lakaiola1128 For developers, the GRBL source code is quite well commented through out making it easy to follow. And because it is open source, everyone can review the source code.

ashelly, 2017-04-13:
Actually, that smoothie diagram is almost a perfect description of how Grbl works (or how it did 2 years ago - the last time I worked on it.)