grbl/grbl #1210: Is there a way to either home X and Y before Z or home Z slower?

CantSayIHave, 2017-04-15:
I have a CNC built from a China kit, it works very well. I'm installing the limit switches, X and Y were easy as they are static. For the Z, I started with a switch above and below the spindle attachment. Then I realized this won't be accurate for homing if I change the drill bit out. I then tried attaching a limit switch to the side of the build plate, which works great, except for when the drill bit is literally anywhere else above the plate.

I had forgotten that Z homes first. At this point, I believe I have two options: make grbl home X and Y first, so that Z always lowers at the same spot, or switch from limit switch probing to electrical probing. If I take the latter otpion, I will definitely need to slow down the Z seek, as at the current speed it would probably crash into the plate and destroy the bit. However, if I do this, I would like *only* the Z seek to be slow, as it would take **forever** to home the X and Y.

Does anyone have suggestions for the best method to take here? I'd like to make this at automated as possible.
Thank you!

jahnj0584, 2017-04-15:
Homing isn't meant for or cares about your bit length. If the bit isn't
safe with Z being raised all the way, then it's time to clean your table
because this is not an issue on anyone's machine but yours.

CantSayIHave, 2017-04-15:
I was talking about setting Z-home to the build plate. Because I know it's a thing people do. But maybe it's not the best way. Should I instead home it at the top and then probe for the bit length?

CantSayIHave, 2017-04-15:
To be perfectly clear I got that idea from [here](https://buildyourcnc.com/PrimeronHomingandLimitSwitches.aspx)

jahnj0584, 2017-04-15:
Yes. Home the machine. Probe Z and jog to X/Y and reset your work zero

chamnit, 2017-04-15:
Read this. https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to

The point of homing is to have a repeatable reference point. You can't with the tip of a tool.

closed 2017-04-15 by chamnit

CantSayIHave, 2017-04-15:
I see you closed this, but I still don't understand how the probing process works - I see mention of using 43.1, but also that that code does not acTualyl cause movement. I've also seen some sources suggest grbl doesn't support tool offsets

chamnit, 2017-04-15:
Check the list of supported gcode by Grbl. There are a lot of old blogs posts that are outdated or just plain wrong. So don't believe everything you read. Here's a link to G43.1. http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g43.1

G43.1 is generally only useful if you need to use more than two tools during a job so you can retain the work coordinate system and have a different G43.1 offset for each tool. The probing cycle just finds the offset. Set the first tool as zero and the next tools are relative to that.

CantSayIHave, 2017-04-15:
So I would have to do the offset calculation with the results of the probe?

Also, still a bit new to this - which command do I use to probe until a limit?

Thank you very much, this is all very helpful!

109JB, 2017-04-15:
No. You simply bring the tool tip to where zero is, or probe, and then enter a G10L2P0Z0. Grbl will automatically calculate the correct offset to make that z position equal to zero in the current work coordinate system. If yo probe and let's just say your probe block is 1 inch tall you would probe and enter G10L2P0Z1. This will compensate for the height of the probe block