grbl/grbl #1211: Z axis homing issue

ReRamp, 2017-04-16:
I just got my cnc up and running and am now working on a homing system. the x and y work fine but when it starts the z acts like it hit the end stop imediatly. I have checked the wiring. I have all the endstops pluged into x+ y+ and z+.

109JB, 2017-04-17:
Most homing issues are related to wiring, or line noise.  If you are using NO switches, I would first check by disconnecting the Z-axis switch at the Arduino and running a homing command. Disconnect at the Arduino because this will get rid of the potential for noise coming through the switch wiring. Run the homing command and if the Z axis starts homing like it should then the problem is in the wiring, or noise.  Be ready to stop the machine if it does work as it could run into the mechanical limits of the machine. The Arduino reset button will work to stop it if the Z-axis starts homing correctly.

Report the results of this test and we can help from there.

ReRamp, 2017-04-17:
Looks like its a noise problem. Did some more tests and i have narrowed it down to that

closed 2017-04-17 by ReRamp