grbl/grbl #1213: Homing Cycle not working?

Colester3213, 2017-04-20:
Dear Experts,

I can't seem to get the homing cycle to run on the Gcode Sender. I have no issue at all jogging the motors through the controller but when I try to run the homing cycle, nothing happens. I'm currently using makerbot limit switches with 3 wires (Red, Black, and Green) and I've tried all wiring configurations. I am using Protoneer's CNC Shield v3 which has dedicated end stop pins. What gets me is that I hooked up a limit switch to the E-Stop pins and it worked exactly as it should. When I use the same wire configurations on the End stop pins, absolutely nothing happens. If any of you have any insight to what's going on, please let me know! 

Thanks!

[image]

vMeph, 2017-04-20:
Did you set $22=1 ?
The limit switches of 3 wire makerbot im assuming is one wire common 1 for NO switch and one for NC switch so depending in what setting you will use you must set $5=0 for NO and $5=1 for NC setting

chamnit, 2017-04-20:
Also read this: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches