grbl/grbl #1214: Porting to ATMEGA64 - PCINT absent

dolence, 2017-04-21:
I have a plenty of  custom atmega64 pcbs laying around, so I decide would be nice to try grbl on them. Unlike newer atmegas, atmega64 doesn't have PCINT on every IO pin. Instead it does have four common INT on PD0...PD4, which should be OK since they are better regarding the fact each INT have it's own interrupt vector and it's easier to track in which way the transition occurred.
I would like to remain as closer as possible to original GRBL and some atmega32 implementations I saw simply aren't using interrupts. Anyone tried something like this? I would appreciate to see how it was done.

Thanks in advance.

chamnit, 2017-04-21:
If you wait about a month, I should have something like that to help this type of problem. If you can't, this [RAMPS port](https://github.com/minsk-hackerspace/lasercutter-grbl-Mega-RAMPS) did a pretty good job at decoupling the same port issue.

closed 2017-04-21 by chamnit

dolence, 2017-04-21:
Doesn't RAMPS use atmega2560 which is very similar to atmega328 in this aspect? I mean, atmega2560 have those PCINT interrupts present on 328. Did I misusderstood something?

chamnit, 2017-04-21:
Sorry misread your post. However, I'm having trouble understanding what you want to do. Grbl only uses external interrupts for hard limit and control input events. Homing and probing are polled. If you don't need either of these, then you don't need external interrupts. Otherwise you'll have to figure out how to route them to these specific pins. You also have to make sure all of hard limits are the same port and the control pins are on all on a different port. This is the way Grbl is written. The link shows you how to use pins on different ports.

dolence, 2017-04-21:
I don't need either at this moment. Will go basic first and address this later. Thanks!

langwadt, 2017-04-21:
with probing being polled a couple of diodes could remove the need for limits/control being on interrupt capable inputs, just diode OR(AND depending on polarity) all the limit and control pins to a interrupt capable pin

dolence, 2017-04-22:
By limit you mean hard limits or homing switches? I'm having some trouble understanding how it's handled. I'm coming from emc2/linux CNC and they generally are different things.