grbl/grbl #1216: V1.1 homing problems

graunchit, 2017-04-24:
I have been using Grbl0.8 laser mode for over a year without problems. I thought I would try v1.1, and now there is a problem with homing that others seem to have had as well.

One axis never reaches the limit switches. The machine goes through the process of homing with the pull offs from the switches ok but it does it with one axis in mid travel.

Tried different Arduino boards with the same result. 
Reload v0.8 and everything works 100% ok again.

Has anybody found a resolution to this?

lalo-uy, 2017-04-24:
Do you use Z axis in Laser mode? from v0.9 on the z limit switch and the
pwm output for the spindle switch pins.

graunchit, 2017-04-24:
No Z axis. The homing is set to just the X & Y axis.
Homing each axis individually makes no difference either, the axis'  still stop short.

Setting the max axis size to larger values has very little effect on the distance for the homing seek travel distance. The calculation for the homing seek distance seems to be going awry.

I am just going through the code for the limits now to try and see what is happening.

chamnit, 2017-04-24:
@graunchit : If Grbl is performing the homing sequence with the seek motion, pull-off, locate motion, pull-off, and end, then its working properly. It's highly likely your problem is related to noise on the limit switch pin. Between Grbl v0.8 and v1.1, there have been a lot of improvements and that includes the homing cycle. So it's not surprising that Grbl v1.1 behaves different. Although, it's curious why you never ran into a noise problem with v0.8. 

Also please provide your $$ Grbl settings and $I build info strings. This will help assess any oddities.

graunchit, 2017-04-24:
I have had to switch back to 0.8 for the moment because I need to use the machine. It is performing 100% without any problems.
I will try a newer version again later on and post details of what happens.

jahnj0584, 2017-04-24:
What GUI are you using to control?

graunchit, 2017-04-24:
For testing, sending commands from a serial terminal,  GtkTerm

mkeyno, 2017-04-29:
@chamnit I have same exactly problem , I've double check the wiring and config file ( `  #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  `), but it seams its just applied on y axis

graunchit, 2017-04-30:
At the moment, my machine is set up and working on Grbl 0.8. I am not going to change it again until I have a few hours spare to experiment with it.

mkeyno, 2017-04-30:
@graunchit would you please share your config file or if have building weblog kindly let me know , my try to use grbl in cnc laser for cutting and engraving so unsuccessful

chamnit, 2017-04-30:
@mkeyno : First, make sure you altering the config.h file inside your Arduino library folder. Not the one you downloaded. The Arduino IDE makes a copy of the folder when you import it.

Second, you are exceeding the max step frequency of 30kHz, as stated on the front page. It's computed by: (step/mm) * (max_rate) / (60 sec/min). You are sitting at 42kHz. Grbl will become unstable above 30Khz and will do strange things. 

If you are wondering why there isn't a check for this, there is, but there isn't much flash space left and is disabled by default. You can enable it in config.h.

closed 2017-04-30 by chamnit

Ishmael2, 2019-02-07:
did anyone have a solution for this
i have the same problem and am not using lazer mode @chamnit please can you share your config.h and how you made yours work