grbl/grbl #1218: homing issue.

micnacc, 2017-04-25:
hello. I'm having isues after homing. the problem is probably not the machine but me. to explain, I home the mache and everything works fine but if I try to run a program the machine runs to the complete opposite side and hit my other limits. ive tried different programs ending in the same result. I'm thinking the machine is homing into a negative space and when I run a program it tries to go back to zero to start the operation. please help. I am very new to this.

chamnit, 2017-04-25:
@micnacc : Grbl sets the machine space as negative. You need to start using work coordinate systems to set your "working coordinate frame" to your work piece or however you prefer to use your machine. Machine coordinates are just the reference coordinate system that is always consistent. See [LinuxCNC's docs](http://linuxcnc.org/docs/html/gcode/coordinates.html) on this topic.

closed 2017-04-25 by chamnit

micnacc, 2017-04-25:
Can you please explain. I am extremely new at this and some things need to
be explained as if your teaching a toddler to do physics thank you

chamnit, 2017-04-25:
@micnacc : Sorry nobody has time for that. It'll require pages upon pages and hours upon hours of time. I suggest that you start by doing your homework. Read the LinuxCNC documentation. There are plenty of other resources online. Google is your friend.