grbl/grbl #1219: Inverted limit switches should not disable pulldown

Harvie, 2017-04-26:
Hi,
i've understood that inverting limit switches using $5=1 is disabling internal pullup. I think this does not make any sense. You should physicaly read switch state same way for NC and NO switches. Only difference is that you should invert how you interpret that state in code.

I don't see how disabling internal pullup helps. Especially when most people go for NC setup because of noise, which can be partialy alleviated by using internal pullup.

chamnit, 2017-04-26:
@Harvie : Please don't double post. It's a waste of time for everyone.

closed 2017-04-26 by chamnit

Harvie, 2017-05-10:
@chamnit sorry, i wasn't sure which repo is active. maybe you can disable issue tracker on inactive forks?

chamnit, 2017-05-10:
@Harvie : No problem, but this site isn't a secondary fork. It's the repo for the older versions of Grbl v0.x. Grbl v1.1+ is at the gnea repository. It also clearly states that this site will be deprecated soon on the front page, as well as on the wiki. For reference, I monitor both sites closely for different reasons. So posting to one is sufficient.