grbl/grbl #1220: Omni wheel robot for unlimited X-Y axis travel distance laser burner

organetic, 2017-04-26:
I'm looking about the possibility to use the fantastic GRBL firmware to act as the brain of a Omniwheel robot.

I believe this would be a fantastic feature to enable grbl on unlimited moves.

[https://www.youtube.com/watch?v=HBdZzmTGeaI](url)

This way, we could even make our GRBL enabled pen plotting or laser burning machines.

Do you believe a new planner is needed? At first sight it seems cartesian will do...

109JB, 2017-04-26:
I don't really have an answer to your question, but a potential problem with using one of these as a plotter or laser machine is that I see them running on carpet mostly, and if you look at the video you linked it looks like a carpet remnant was placed specifically for that test. Thinking about it, if the traction of one wheel is less than the other the machine could rotate, throwing off any alignment for a plotter or laser type machine. My guess is that the carpet is to more closely match the traction of each wheel. On a hard surface I doubt the alignment would remain stable enough for good results in a laser or plotter application.

chamnit, 2017-04-26:
@organetic : Aside from the technical issue of Grbl supporting 3-axis (motors) and your robot having four (and maybe more), I don't see that much of a restriction of modifying Grbl to run your robot. Adding an axes or two is fairly trivial if you use an Arduino Mega or chip with more flash and RAM.

However, keep in mind that Grbl (and most CNC firmwares) assumes axes are orthogonal or not dynamically coupled. If your application is ok with that, then it should run good enough to get going before figuring out the optimized control algorithms. If you're curious, non-orthogonal axes moves into the realm of robotics and large sets of equations that take quite a bit of CPU power to solve in real time. Add a motion control path algorithm, then the problem blows up quite quickly. 

Grbl doesn't really care about unlimited moves. As long as you continually stream new motions of short length and duration, just long enough to keep the planner buffer full and the robot can maintain its full speed. See the jogging wiki to see how to do that. This way you can alter the robot trajectory on the fly with low latency while still computing the motion plan within acceleration and speed limits.

closed 2017-04-26 by chamnit

organetic, 2017-04-27:
My first thought was to use 4 nema 17 stepper motors, one on each wheel, with Y and X axis mapped to each pair. 

Just like this picture:

[image]

So, maybe there's even no need for more axis.

Regarding grip/traction, there's a video showing the behaviour on tiled floor. 

[https://www.youtube.com/watch?v=RSmo1JfEsXU](https://www.youtube.com/watch?v=RSmo1JfEsXU)

So, maybe using a heavy chassis, and keeping acceleration and speed in lower values can work.

What do you think?

109JB, 2017-04-27:
I'm still not sure I would use it, but why couldn't you just instead of using it like this:

[image]

use it like this:

[image]

organetic, 2017-04-28:
@109JB Unless I'm seeing it the wrong way, the latest youtube video that I have linked shows a robot that works exactly the way you recommend...

okaygisiz, 2018-05-11:
https://www.youtube.com/watch?v=YsNdAuhQeFg

beckmx, 2019-03-19:
@okaygisiz what grbl version did he use?

DAguirreAg, 2019-08-06:
Helo everyone! did anybody try building and testing the grbl on a omniwheel robot? Currently I am trying to build my own robot (3 omni wheels) and I would like to use grbl. However, I am in the stage of modifying grbl code, but I am not sure which file/files to modify.