grbl/grbl #1222: Suggestions for vinyl cutter

dolence, 2017-04-30:
Hi!

I'm using an STM32 GRBL 1.1 port to drive a homemade vinyl cutter made from parts of an older dot matrix printer. I saw some others doing the same, mainly using LinuxCNC or Mach3. I would like to suggest a vinyl cutter mode or at least some help on how to acomplish those things by myself. Any help would be appreciated.

**Automatic home search on power up or reset:**
Since I wanted a seamless use, this feature would be really nice. I adressed this by calling   mc_homing_cycle to the begining of system_execute_startup(), but I'm not sure that this was the best way to deal with it. I need X axis only to be homed.

**Axis independent soft limits**
While it's very nice to have soft limits on X, such feature can't be applied to a specifc axis only. My Y is responsible for the paper/vinyl traction and doesn't need to be homed. Once the material is on position it should scroll infinitely, without limits. Setting a very large maximum travel to Y and somehow auto zeroing Y axis before every job start would do it.

Thanks in advance.

sbkivitz, 2017-05-01:
Hi Dolence,
I'm interested in your STM32 port and your vinyl cutter machine design. Do you have any details online? I didn't see the fork in your github account..

Thanks,
Scott

chamnit, 2017-05-01:
@dolence : Automatic homing is not recommended for general purpose CNC. For your case, it seems somewhat ok, since it's a light duty machine that won't likely injure anyone. If you do it anyway, I would just make it EXPLICITLY clear that your Grbl fork does this. Automatic homing can lead to some bad crashes when a user isn't expecting it to happen. It's always better to have the GUI do this for you, since it's trivial to just have it send a command, if you have your own custom one.

As for axis independent soft limits, I'll add this to my notes for a future feature. Got my hands full for now, but this is a good idea. Thanks for sharing.

The code is well structured. Look at the main loop in protocol.c to do something prior to entering that loop. Look in limit.c for the soft limit check routine. It's easy to alter for your purpose.

closed 2017-05-01 by chamnit

dolence, 2017-05-03:
@sbkivitz I think you misunderstood me, I'm not the author. 
Here is the link for the port: https://github.com/usbcnc/grbl

@chamnit I guess I will take another route. I will let GRBL/STM32 almost untouched and wire it to another microcontroller. I should be able to home as desired, stream files from SD, USB or Wifi, etc. Do you have any plans for an oficial port to STM32? Those boards used by usbcnc/grbl are very cheap and powerful.

usbcnc, 2017-05-03:
@dolence ,STM32 should be powerful enough to stream files from SD card as well as display it on an LCD display and drive a keyboard. Maybe slightly higher end of chip to have more IO. But the board itself should be still relative cheap.

chamnit, 2017-05-03:
I've never been a fan of sharing real-time tasks with other processes. I don't recommend altering Grbl to add SD cards, LCDs, etc. You'll likely find yourself constantly running into problems unless you understand the Grbl code very well. It's always better to interface it with something separate and since an STM board is cheap, it won't cost you much to add another one. It'll also save you a huge headache in the long run.

dolence, 2017-05-03:
@chamnit I totally agree with you. Do you have any plans for an oficial port to stm32 on a near future?