grbl/grbl #1223: Linear Encoder Closed Feedback Loop

schilly91, 2017-05-05:
Hi,

I know, that there have already been some discussion about grbl and linear encoders and @chamnit set it to his To-Do list.
So I would like to integrate some linear encoders (glass scales for each axis) in grbl. I have adjusted the grbl project to a Cypress PSoC 5 controller like @bdring it did and I already get a feedback from the linear encoders in mm (resolution 1µm). For my question the hardware doesn't matter.

There are different ways to integrate it, but I do not have the right idea yet. 
I was thinking about to integrate it directly in the stepper ISR or to write an new interrupt which is doing the correction. 

 **Does anyone has a good idea how to do that?**

Thank you for any hint.

chamnit, 2017-05-05:
@schilly91 : It entirely depends on what you intend to do with the linear encoder data. Do you just want a DRO? Do you want closed-loop feedback with a simple notification of position loss? Do you want to use that data to perform positional corrects on the fly? Are you controlling a servo, rather than a stepper? All of these things matter and each rapidly increase in complexity.

schilly91, 2017-05-05:
@chamnit: Thank you for your answer. 

I already have DRO at the moment. **So I would like to implement a closed-loop feedback with positional corrects on the fly.** I'm using steppers and would like to get a 1µm accuracy. My setup can provide very small axis movements (theoretical ~10nm per microstep), it is for a very precision positioning. 
There is some axis backlash (depends on the axis, ~5µm). The idea was to compensate this by integrating the existing linear encoders.

chamnit, 2017-05-05:
@schilly91 : That isn't something very easy to do. First you have make some decisions based on your intended purpose. How will it compute and control a correction, how quickly, do you have a large mass (inertia) to deal with, does it matter how it does a correction, etc, etc. If you want my opinion, it's much easier to attach a separate controller between your motor drivers and Grbl. It's job is to read the linear encoders and perform positional corrections based on the step/dir count it has received. This way you can customize it freely without having to alter any of Grbl's code and break something.

schilly91, 2017-05-06:
@chamnit : I was searching for a way to integrate it in grbl, but as you said it's not very easy. I would also say to see grbl as a blackbox is easier. 

The idea to count the steps and compare it with the linear encoders sounds good. I would like to run it on the same microcontroller board, because I still have enough resources. 
What I think to do is a separate C module with my code. The input would be the step/dir signal, compute it to mm and give it to a closed feedback loop with the measurement of the linear encoders, where I compute and regulate my new step/dir signal. **I think that's the way how you mean?**

duronebis, 2017-05-06:
May be a chasing pattern (similar to the way the Cnc included in Theremino works) could make things simpler. 
In this way, instead of precalculating a deterministic motion control, you should just set the right step/dir to reach the 'goal point' starting from the point you got from the encoder. Then the goal point must always move with a time lap in synchro with the Feed you need.
Theremino approach works in this way, but it starts from the current supposed position, not the real one.
Just a thought.

chamnit, 2017-05-06:
@schilly91 : Sure, you can integrate it in the same manner as suggested with the separate MCU. You link the stepper ISR to another module that computes the correction factors and alters the timing. However, this can get very processor intensive and can break things if you run out of processor performance ceiling in certain scenarios. You'l have to come up with a very lightweight implementation that can run real-time.

@duronebis : The whole point of CNC is deterministic motion control. You have to follow a path explicitly and as fast as possible. However, you can integrate a "chasing pattern" with a control algorithm to ensure the machine is on the right path and to improve accuracy. This is basically how servo-based CNCs work. The drawback is you have to set aside a large torque/performance margin to be able to execute those correction factors. How much is dependent on how quickly you can compute and command those corrections. The faster you can, the less jitter you introduce into the system and you have less inertial force you have to fight against.

MarcusWIA, 2018-04-19:
@schilly91: Did you manage to implement the encoder into grbl as you were trying to do?

I would really like an opinion on that because I am researching for my Diploma Project and I need a closed loop on at least one of the steppers of my CNC.

wolfgangr, 2024-08-02:
Hi out there,  

Any news on this issue?  
... and sorry for necromancing, please ...   

I'm on the way to resurrect the topic out there:
["Long closed loop" from linear encoder into grbl stepper for dynamic backlash compensation #563](https://github.com)/grblHAL/core/discussions/563