grbl/grbl #1225: Understanding Grbl

tracy1996, 2017-05-09:
I have read somewhere on github that there is no bit left for anymore code in grbl. how far this is true. How can i add more code to grbl for development?

chamnit, 2017-05-09:
There's about a hundred bytes left for certain critical build packages. From default, you have about 1KB of flash and zero RAM (reduce buffer sizes to free some up). Otherwise, wait for the abstracted version that is portable or use the Mega2569 version. See the Gnea repos.

closed 2017-05-09 by chamnit

tracy1996, 2017-05-10:
hi chammit, thanks for rapid response :) what are gnea repos? i have a mega 2560 lying around will that work?

chamnit, 2017-05-10:
Mega2560. A typo. Look [here](https://github.com/gnea/grbl-Mega/)

tracy1996, 2017-05-10:
chammit thank you for the link. so if i add code to this it wont effect the performance?
please correct me if i am wrong.

chamnit, 2017-05-10:
Of course it will. How much depends on what you are trying to do, how efficiently you code, and how much math is required. If you write it to be modular, you'll likely be able to use it in the ARM version that will be out soon. Until then, it's the Mega2560 branch or some other unsupported ARM fork of Grbl.

tracy1996, 2017-05-10:
so how do i start? can you give me a little guide line. will i only have to make a new cpp and h file and include this file in main.cpp? and compile the code again? a little confused here. i would like to test out an lcd driven directly from mega.

tracy1996, 2017-05-10:
and secondly  whats the hitch with speed? whats eating so much speed in it :) ?

tracy1996, 2017-05-12:
anyone there?

chamnit, 2017-05-12:
@tracy1996 : I don't have the time to walk you through everything, but to get started I suggest you learn how to compile Grbl. There are a few methods laid out [here](https://github.com/gnea/grbl/wiki/Compiling-Grbl). 

Speed limitations are due to the Arduino processor not being that fast. Grbl has to make a lot of calculations and maintain them in real-time. Any significant math or a blocking task that takes too long can cause Grbl to crash. Start small and add things slowly.

tracy1996, 2017-05-12:
thats rude :(

Protoneer, 2017-05-12:
No its not. ;)

109JB, 2017-05-12:
Rude?  @chamnit uses HIS personal time to maintain and improve this wonderful thing we have called Grbl.  He doesn't get paid for it and what little time he probably has should definitely not be spent hand holding someone so that they don't have to do research into how Grbl works. There is a lot in the works that will benefit everyone rather than one-off projects. He also has a rjob he has to do to make a living. Countless people have benefited from @chamnit work he has put into Grbl.

chamnit, 2017-05-12:
@tracy1996 : I've told you everything you need to know to get started. The source code is very well commented. You will have to do your own homework. You'll quickly learn that this is common for open-source projects, because they often don't have massive support budgets to help users. So please be respectful.

gerritv, 2017-05-12:
The other answers are too gentle, not rude! This has nothing to do with open source, it is just requires some work to read the code. No project, open source or commercial is going to hand hold you while you learn.
If you ask a question such as 'so if i add code to this it wont effect the performance' then you are very obviously not ready to do the level of programming and debugging required.
Since you want to learn to write code, I highly recommend working on desktop apps before getting into this type of project where you can't debug in any meaningful way.

electrokean, 2017-05-12:
@tracy1996 spend some of your spare time reading through the source code, and more importantly all the discussion on the closed issues here (and on the newer version at https://github.com/gnea/grbl/) and you'll learn a lot more than anyone could write here in a few paragraphs.  You'll then be able to answer your own questions, and maybe even contribute back to the project.

tracy1996, 2017-05-12:
Chammit i am sorry if you mind that. It wasn't intended in -ve context. Your work has inspired me to join this awesome community. 
I love to read through the code but there are certain question which are confusing me. I just meant to have a good quick start to contribute. 

Thank you electrokean for a kind response i definitely have a look. 

"Unity is strength... when there is teamwork and collaboration, wonderful things can be achieved."