grbl/grbl #1227: CNS chield v3.0 and spindle relay

renat2985, 2017-05-10:
I'm using the latest firmware GBRL 1.1

I connect relay in this way:
[image]

The relay does not turn on. Please tell me why?

jahnj0584, 2017-05-10:
The shield is made for GRBL 0.8. The spindle port changed in 0.9 onwards

renat2985, 2017-05-10:
Can you tell on which pin I can now connect the relay in my shield?

vMeph, 2017-05-10:
that shield is for version grbl 0.8, on version grbl 0.9 and above there was a change between pins 11 and 12, you can adapt that shield to work with grbl 0.9 or 1.1 you will need to swap pins, one way to do that is you bend the pins 11 and 12 on the shield and then you put cables with female header attach to them and solder the other side on the bottom of the arduino, so pin 11 of shield you solder to 12 of the arduino and the 12 of the shield you solder to the 11 of the arduino

you can see in this page the changes that was made from version 0.8 pinout and the version 0.9 pinout
https://github.com/grbl/grbl/wiki/Connecting-Grbl

stefrafr, 2018-06-08:
hi renat ,i don't know about you but i had the same problem.After i swap the pins in cnc shield nothing happens until a change the grbl config in $30 from 1000 to 255.I am using vectric aspire for making gcode and the J TECH GRBL NoZV9(mm) for post proccesor and works great. Hope to hepl you

McKlay, 2019-10-20:
Thank you so much guys i see there's no helpful guides in youtube for this

changos-gt, 2020-01-03:
**I will describe in detail how I got it working**. Built a few laser cutters a while ago using Arduino and the CNC Shield, and I'm building a new one, same hardware. Yes there is something weird going on with the setup. While I will describe in detail what I did (at the end of the post), I won't cover in same detail the many variants of the problem given the different versions of the shield and GRBL software (for Arduino) considering the many supposed "hacks" (that actually don't work or do so partially).

**First: how this is supposed/expected to work**. The CNC software and shield should turn on and off the laser getting on and off signals from the respective pins. The wiring should take place using a RELAY connected to + and -, and the third cable to the signal pin (SPN-EN), yes, it should be that easy on simple laser setups, but it doesn't work that way... exactly. Now keep in mind there are diff versions of the CNC shield and there are several diff versions of GRBL for Arduino, you will easily find lots of different forum posts on the problem and diff supposed solutions depending on the version of each of those, along with many comments stating "_hey, this doesn't work_", because there are pin changes made on diff GRBL versions, so long story: it's a big salad. I won't go into details due to the variety of setups possible in this context.

Before posting what I did (that works for me as expected), let's talk about two possible solutions.

1. Using an Arduino Relay Module, this thing has 3 cables, +/- and signal. The problem here is things work opposed as they should, meaning when you turn OFF the laser via software controller: the relay gets activated; and when you turn on the laser, the relay gets deactivated. Well, a relay having 3 connections (Open and Close) means you could take the easy path and connect your laser the opposite way, meaning it will get power when the relay is off, that should do it, right? the problem with this is the laser will turn on when you connect your machine (about 3 seconds) and then it will go off, well that's a small price to pay but that's not the best approach, it's just not safe. Yes, it sucks that the laser will get activated when the pin doesn't send a signal, it's all upside down.

2. Use a transistor to reverse the logic. You could control your laser with the transistor using reverse logic, or you could use the transistor along with the relay and control your laser in order to have "laser on = laser on", simple as that.

I said two possible approaches but there are more, those two are the ones that sound best for me (you could google the software editing hack on the GRBL) but I didn't like the other options because they involve more changes, I like simple things like just updating the GRBL any time I want to (without worrying on edited code) so you could google that yourself. But let's be clear on something: yes, the initial problem is the laser would work the opposite way as expected, that's it in general and if you take the time to search: you will find lots of forum discussions and complains, including the supposed method of twisting or changing some pins (hardware).

**Now what I did and works perfectly for me**: (details after the brief explanation)
I'm using a low powered relay directly connected to the signal pin and GND on the CNC shield, yes it works. The relay turns on when I send the "turn on" signal, and so my laser works as expected. Yes, I'm NOT experiencing any laser-on when connecting the machine, so this is a very safe approach in general. It works as expected. :+1: 

:/ Yes I can see some people being happy with this approach and some others unhappy, complaining about this not being the right way to do it, yes you shouldn't connect a relay directly to the signal pin, agreed, why? 

Most relays need too much current to work, so you might face wiring things and not getting the relay activated due to not having enough current, plus, if you get it activated... it might abuse the current limits of the Arduino output causing unwanted issues like heating, destroying the pin or overall failure because that pin is taking power away from the other Arduino functions (there is detailed information about this if you know how to search for it, I'm not making this up). I just wanted to explain details around the solution I'm using, and only about this solution, you can research the rest yourself.

** :+1:  Details of what I'm using and why it works, why it is safe given my setup**: Well, I'm using a low power relay. It consumes 30mah-35mah max (measured and confirmed), that's on the safe range of what an Arduino pin can handle, you should stay BELOW 40mah and you can research about that. The relay MUST use a flywheel diode, otherwise you will eventually fry your Arduino. Coils, relays, etc will use energy (and store it) so when you turn off the power, a big spike of current and voltage will flow back on the same line, this means you can get easily more than 10v back to your Arduino when you power off the relay. A diode in parallel (in the right polarity) will kill any current wanting to go back on the power lines.

That's it, it works for me, I've had my last machine running for almost two years and as said I'm building the next one.

...But wait... what about EMI? electromagnetic interference? before considering my approach being problematic keep in mind to consider religiously what really causes EMI and how to deal with it on your machine, so this  [... cut ...]
If you still want to discuss EMI and complain about it, consider reading about the different and best cable options, in short the best way is to use shielded twisted pairs along shielded twisted pairs inside shielded cables (yes as you read, shield inside shield). There are good videos about that on YT.

That's it, if you reader don't like the approach that's fine, I described as much detail as possible, there are good readings describing why this is safe (under the detailed described circumstances).

Good luck.

* Decided to post here because it's the first relevant search results appearing on Google, that should save some time.