grbl/grbl #1228: Arduino Upload error

rinaldoj, 2017-05-12:
We are trying to upload grbl libraries without success. 

Error when is writing (avrdude: stk500_recv(): programmer is not responding)

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 3.3
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\AppData\Local\Temp\arduino_build_319898/grblUpload.ino.hex"
avrdude: writing flash (30058 bytes):

Writing | ###############################################avrdude: stk500_recv(): programmer is not responding

gerritv, 2017-05-12:
The boot loader on your Arduino is taking up too much space. You will need to flash a smaller one (Uno style) or recompile Grbl to remove code, such as enabling GUI mode (no strings for error messages etc).
I would suggest you also use Gnea 1.1 as it is being supported. (https://github.com/gnea/grbl)
There are several Closed Issues here that describe the boot loader flashing process and where to get a smaller bootloader.

chamnit, 2017-05-12:
30058 bytes is below the 30.5KB (31262 bytes) max for non-Arduino Uno boot loaders. But I do agree that you likely need to re-burn your boot loader. I've run into this issue in the past and re-burning it usually fixes it.

closed 2017-05-12 by chamnit