grbl/grbl #1229: Cycle Start Does not start the program

Biggs427, 2017-05-12:
I'm running GRBL 1.1 on  UNO R3 and is the GUI.

I built a pendant hooked to the A0 (Reset), A1 (Hold) and A2 (Start / Resume) pins. 

Everything works well except that the Start/Resume button is in fact a Resume.  

When a program runs, and I hit the hold button, the programs pause until I hit the resume button; either software or hardware.

But if a program is loaded and I hit the hardware Start/Resume button, it doesn't start. 

Thanks

chamnit, 2017-05-12:
@Biggs427 : It used to be, but no one ever used it that way and just added an extra step. I recently altered that behavior in v0.9+ and the stream now controls how a job starts. This is fine because Grbl doesn't know whether the gcode coming in is part of a program or not. It's the job of the GUI to deal with that instead. But, you're right. There is some documentation that needs to be updated to get rid of this minor mislabeling.

closed 2017-05-12 by chamnit

Biggs427, 2017-05-12:
Is there a way to reactivate this via the config.h?

Thanks

electrokean, 2017-05-13:
@Biggs427 Well you'd need to separately start streaming your gcode first anyway, so maybe including an M0 (pause) at the start of your gcode will achieve what you want?

chamnit, 2017-05-15:
@electrokean : Hmm. That's an interesting idea. Over the years, I've always wanted to figure out how to handle the `%` character that most g-code programs start with, so to mimic cycle start behavior. Interpreting that as a `M0` pause internally would likely do the trick. I'll add this to my development notes as a potential solution.

einencool, 2018-06-28:
Hello,
Would it be possible to change the function of the start_cycle button, that it will really start the job?
I‘ve got the hardware buttons right in front of the machine. The PC is a few steps away. Sometimes I need to make 30 pieces and then it would be great to start the job with a single press on the „start button“.
Greets 
Chris

cri-s, 2018-06-28:
Start the program with
!G4P0

and run the program.
This allows you to start the program from hardware button,

einencool, 2018-06-28:
Well, thats sounds a little bit annoying.
Then I have to click 30 times on the software button to start the job AND 30 times on the hardware button to resume the job.... Is it right?

Somewhere in the code there should be the possibiliy to change the command from a „!“ to a „~“, am I right?

cri-s, 2018-06-28:
It depends on the sender program, but it's the same as M0
Supposing your sender program intercept M30 and change it to M2 rewinding
the code and starting it again. If you code already contains one M0,
Yes, if you make 30 pieces, you have to press the start button 60 times in
this case.
I prefer the !G4P0 because then i know it's GRBL fix code and different to
M0 present to the program for other reasons.

langwadt, 2018-06-28:
@einencool  The sender is the "master", GRBL can't tell the sender to start sending data it just consumes the data it received so the GUI can send it more data. You 'll always have to tell the sender
to start sending data

einencool, 2018-06-29:
Ok, thanks for the explanation. So I thought the description "Start cycle" means that I can start the program. So for my opinion the hardware button is very useless.

But thank you all for your fast replies.
Greets Chris