grbl/grbl #1230: something is wrong  with safety swuches , ALARME: Homing fail

cuchara, 2017-05-14:
Hello everyone

I just finished to installing the safety swuches, And I activate the parameters: 
```
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)

```
But when I send the command $H , the Z axis begins to rise  antil it touches the safety swich, the axis stops
Console:
```
Grbl 0.9j [`$´for help]
[`$H´l`$ X´ to unlock]
ALARME: Homing fail
ok
```

No matter what I do, message repeated and I do not regain control of the machine. till I turn off and disconnect everything
Then I move the  Z axis with the hand to unlock the swich then I connect everything back

Console:
```
Grbl 0.9j [`$´for help]
[`$H´l`$ X´ to unlock]
ALARME: Homing fail
>>>>44
error:Alarm lock
```
And I restore the control when I turn off the alarm $X   

Try the homing many times and touched the switch with my finger to see what happens and nothing happens just stops but if I loose the swich it continues with its path till it blocks itself

What is wrong?

[image]

vMeph, 2017-05-14:
to do only homing try with out soft limits and hard limits at 0 and do homing see what happens might work but for the hard limits sound like you have noise issues in your switches that will trigger a alarm, you might want to take a look at 
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

jahnj0584, 2017-05-14:
Sounds like it's wired backwards. Use a volt meter to test the continuity
and make sure it's behaving properly

cuchara, 2017-05-14:
Hello
Thank you very much for the help
This is the way I connect the swiches https://okelectronic.files.wordpress.com/2015/07/mycnc_connections.jpg

Then what is the appropriate form or which do you recommend?

  I do not have the

cuchara, 2017-05-14:
> to do only homing try with out soft limits and hard limits at 0 and do homing see what happens might work<

I did what you recommend and  happens the exact same thing

`>>>$H
ALARME: Homing fail`

vMeph, 2017-05-14:
first make sure all switches wired correctly do a test like  @jahnj0584 suggested if all is ok
most like you have a noise problem witch is very common problem around here to fix noise issue problems there is many things you can do like shielded cables to you motors, proper groundings, and add pullp up resistors and capacitor to filter noise on switches read the link i sent you before is there alot information in how to approach this problem

chamnit, 2017-05-14:
Soft limits and hard limits should not effect homing. Both are disabled when it runs. There are only a few things that would throw a homing fail alarm. Use Grbl v1.1 instead. It has better error codes to tell exactly what failed. Link to dv1.1 site on front page.

cuchara, 2017-05-14:
That sounds good...  because at the moment I do not have capacitors . 
I'm going to try the Grbl v1.1 

  Thank.👍

cuchara, 2017-08-28:
I already passed to GRBL V1.1  And my problems are other 

 I not only change the Grbl v.1.1  but also I change the steppers drivers  and I do not seem to achieve an optimal configuration,  I try changing everything

My motors Nema32
Specification:
For Motor Length 45mm:
Rotor Inertia
180 g.cm2
Step Angle
1,8°
Motor Length
45mm
Holding Torque
0.8N.m
Rated Current/phase
2A
Phase Resistance
2.1Ω±10%
Rated Voltage
4,2V DC
Inductance/Phase
5mH±20%(1KHz)

My drivers 4A TB6600
http://www.ebay.com/itm/Newest-4A-TB6600-Single-Axis-Stepper-Motor-Driver-Controller-for-57BYG250H-Motor/162217753430?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
I have them set to:  

MicroStep  2/B  400

Current(A)  2.0

My  parameters
[image]
[image]
 The most common problem that occurs to me is that when the machine is cutting, automatically enter in  into pause and it automatically relaunches
 
Also when I change the speed of the spindle  manually grbl is enter into pause

But the one that puts in pause is not serious that bad...The worst problem is when he sends me error this is what is driving me crazy I have not been able to cut a piece without error. And to solve the error I have to restart everything from zero

[image]

 Which means that I lose the position of the machine and it is very difficult to find the point 0 again manually

 This cost me many badly cut pieces and carbon plates in the trash
 What can it be?
 For now I have not tried the safety limits

chamnit, 2017-08-28:
Something is triggering a soft-reset command mid-job, based on your `Alarm:3` code. Given you are running large NEMA32s, this likely means that there is electrical noise effecting the soft-reset pin A0. Place an RC filter on the A0 soft reset and any other control pins you have connected.

closed 2017-08-28 by chamnit

cuchara, 2017-09-04:
Can you please give me an example of how to add an RC filter to the pins?
thanks in advance

lalo-uy, 2017-09-05:
Check this

https://www.allaboutcircuits.com/uploads/articles/filtered-signal-v2.jpg

4k7 Rpull-up, 100 ohms serie R and 100nF cap will do

cuchara, 2017-09-05:
mmmmm : /     so something like this ? 3
https://static.xx.fbcdn.net/rsrc.php/v3/y4/r/-PAXP-deijE.gif

lalo-uy, 2017-09-05:
Just see a 1 pixel gif

cuchara, 2017-09-05:
Sorry
  here it is:
https://scontent-cdt1-1.xx.fbcdn.net/v/t1.0-9/21369429_10154963293221239_603982951625439924_n.jpg?oh=48975306d4b97636cd716349031f02e9&oe=5A549501

lalo-uy, 2017-09-05:
> Sorry
>   here it is:
> https://scontent-cdt1-1.xx.fbcdn.net/v/t1.0-9/21369429_10154963293221239_
> 603982951625439924_n.jpg?oh=48975306d4b97636cd716349031f02e9&oe=5A549501

The pullup resistor is more important than the series one.

cuchara, 2017-09-05:
The two are completely unfamiliar for my, I only attempt to understand how I have to connect the first diagram that you showed me

so can you explain it to me??

vMeph, 2017-09-05:
@cuchara  take a look in wiki schematics for switches wiring to avoid eletrical noise
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

vMeph, 2017-09-05:
based on your diagram you want to do on all switches how you see on the Z limit
 thats a RC filter
[image]

cuchara, 2017-09-08:
Many thanks for your support guys!I  greatly appreciate it.

 I still have questions :D 

I do not have a capacitor like the one they show me, but I have many others

-25v 10uf
-E103Z IKV
-102K IKV 88
-10V100uf
-16V 220uf
-16v 470uf
etc....

which one can I use as a replacement?

many thanks

lalo-uy, 2017-09-08:
103 is the nears, but is 10 x less.
find some 104  = 100nF