grbl/grbl #1232: porting GRBL on STM32

mkeyno, 2017-05-18:
I run couple of Arduino project with Arduino port of [STM32 by Roger Clarc ](https://github.com/rogerclarkmelbourne/Arduino_STM32), his repo is very mature and stable today's and the [STM32F103 ](http://www.rogerclark.net/stm32f103-and-maple-maple-mini-with-arduino-1-5-x-ide/)one of the cheapest board as well as Arduino mini whereas it is more powerful than Arduino mini with lot of ram, I found couple of repo like @langwadt & @MaxwellXyao that port GRBL on STM32 but those mostly old and not inserted new feature of GRBL1.1 , has any official effort  to porting GRBL on STM32

Elvenblood, 2017-05-30:
Hello,

I've been running usbcnc's port for the last day or so. It has some glitches which I hope will be addressed soon. My steppers run supersmooth with this port. However I haven't got homeing to work.

Here's the link:
https://github.com/usbcnc/grbl

I hope more people will use this since it's a good and cheap alternative.

/L

chamnit, 2017-05-30:
@mkeyno : I've already answered this porting question several times. Closing.

closed 2017-05-30 by chamnit

X3msnake, 2017-05-30:
@chamnit 

If you want I can make a push to add recurrent questions like these to the GRBL wiki FAQ

chamnit, 2017-05-30:
@X3msnake : Thanks. That would be greatly appreciated. Place it on the /gnea/grbl wiki, rather than here, since this site is deprecated.

X3msnake, 2017-05-30:
Ok will do

X3msnake, 2017-05-30:
@chamnit

ok, just added this issue to the FAQ,

Hope i quoted you right.

2017-05-30 15:58 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> Ok will do
>
>
>> @X3msnake <https://github.com/x3msnake> : Thanks. That would be greatly
>> appreciated. Place it on the /gnea/grbl wiki, rather than here, since this
>> site is deprecated.
>>
>> <https://github.com/grbl/grbl/issues/1232#issuecomment-304904258>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva