grbl/grbl #1233: cant compile grbl

algo1980, 2017-05-25:
I am having a problem in compiling grbl after adding 2 new files lcd.h and lcd.cpp. when i compile it says recipie target for grbl.elf failed. 

how can i solve this bug?

electrokean, 2017-05-25:
Again, you haven't provided enough information... any "bug" is obviously in your changes, so you need to learn from what we said previously and show us exactly what you've done - or we're just guessing!

Anyway, I would guess that your additional code is too big to fit in the ATmega328 flash memory.  It was already very tight.  I suggest you try working with https://github.com/gnea/grbl-Mega which is newer and runs on the larger Arduino Mega board with more pins and memory, and thus is much more suitable for expansion.

algo1980, 2017-05-25:
Elec i am already using mega. But the problem is i want to add two new files lcd.h and lcd.cpp 
they work fine if compiled separately but when i add them to grbl folder it gives me an error somewhere in the makefile. do i have to make any changes to the make file also?

electrokean, 2017-05-25:
Well then, show us the code and the error.  How do you expect us to help you otherwise?
And you should be using the newer code for the Mega I linked above - this code here is getting old.

algo1980, 2017-05-25:
#include <avr/io.h>

#include "lcd.h"

void t6963_init(void) {

    /* set I/O */
    LCD_NCE_SET_OUTPUT_HIGH();
    LCD_NWR_SET_OUTPUT_HIGH();
    LCD_NRD_SET_OUTPUT_HIGH();
    LCD_CND_SET_OUTPUT();
    LCD_DB_SET_OUTPUT();
    
#ifdef LCD_NPWR_SET_OUTPUT_LOW   
    /* power on */
    LCD_NPWR_SET_OUTPUT_LOW();
#endif
  
    t6963_status_check_mask = T6963_STATUS_CHECK_COMMAND_DATA;
    
    t6963_send_command_uint16(T6963_SET_TEXT_HOME_ADDRESS,T6963_TEXT_HOME_ADDRESS);
    t6963_send_command_columns(T6963_SET_TEXT_AREA);
    t6963_send_command_uint16(T6963_SET_GRAPHIC_HOME_ADDRESS,T6963_GRAPHIC_HOME_ADDRESS);
    t6963_send_command_columns(T6963_SET_GRAPHIC_AREA);
    t6963_send_command_uint8_nul(T6963_SET_OFFSET_REGISTER,T6963_EXTERNAL_CG_HOME_ADDRESS>>11);

    t6963_send_command(T6963_OR_MODE|T6963_INTERNAL_CG_ROM_MODE);
    t6963_send_command(T6963_GRAPHIC_OFF|T6963_TEXT_ON|T6963_CURSOR_OFF);
    
    /* Clear display */
    {
	unsigned i;
	t6963_home();
	t6963_set_data_auto_write();
	for (i=0; i<LCD_LINES; i++) t6963_erase_line();
	t6963_auto_reset();
    }

    /* return home */        
    t6963_home();
}