grbl/grbl #1234: GRBL freezes when I turn on Hard Switches

norbigazda98, 2017-05-25:
Hi!
I have a problem with my GRBL driven Arduino Uno and the three TB6600 drivers, I use. 
I attached the connection with to this post.

So, when I used it only with the drivers, no limit switches, and no finished enable pin, it worked like charm, altough with v0.9j grbl, it froze pretty often, but that almost ceased when I updated to grbl v1.1f. 
When I tried to complete the enable pin with connecting all the enable (-ENA//-EN//) sockets to the Arduino's (D)8 pin, it didn't even boot up completely, it just froze. 
So I reverted back to not having an enable pin connected, like it is on the picture.

The problem
Today I installed limit switches across the whole CNC router, the switches worked, for sure, I tested them with a multiemeter after the whole process on the end that goes to the Arduino board. (I tried all combinations of connecting them serial or parallel and reversing or leaving alone the limit pins (in the GRBL settings, around $15)). 

My question is, what do you think might be my problem? 
Why does grbl boot frozen if I enable hard limits? Can it be in connection with it freezing when I was tríing with the enable cables? 

A saw different connections with the TB6600. Should I try to change all the connections, so that the 2-7 pins of the board would go to the driver's (something+) sockets and the (something-) sockets would be all connected to the GND pin on the board? 
Or do you recommend something else? 

[image]

109JB, 2017-05-25:
A few questions:

Did you use NO or NC switches? If NC, did you invert the limit pins?

Did you also enable homing?  The default is with homing enabled, Grbl will start in an alarm state.

electrokean, 2017-05-26:
What is your $4 setting?
See https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#4---step-enable-invert-bool
Well, you should really list all your settings

norbigazda98, 2017-05-26:
I tried both NC and NO switches with both enabled and disabled reversing, the rasults are all the same.
There is an option in SourceRabbit to kill alarm, but it doesn't do much. 

I didn't enable homing, I wouldn't like it to go around the whole workstation with the tool or the material in it and break everíthing. Yes, I might place the router bits and the workpiece in later, but that doesn't seem to be a good idea, since the stepper drivers are always enabled, because when I try to connect it, it freezes all over me. 

So I'll post the settings and try the homing command, but should I try to check the other type of wiring I asked about in the original post?

norbigazda98, 2017-05-26:
Here are my settings.
I tried splitting the GND pin instead of the +5V pin, but nothing's changed, only the directions are reversed. 

The funny thing is, that it works perfectly with the switches, until I switch on the power supply for the drivers. I tested the limit switches, gave the commands, all worked, when the drivers were without power, but when I give them the 36V, grbl freezes, but only if the hard limits are enabled 😕 
[image]

norbigazda98, 2017-05-26:
Well, I managed to get some results. 
_BTW, the homing command instantly freezes it, even if the power supply is turned on or off._ 

My result is... **interference**... 
I disconnected everything what I connected yesterday to the limit switches and I only had the connector outputs on the outer side of the box of the electronics. I connected the enable pin, and it (the enable pin)  works fine, looks like it needed the reversing of the +5V and GND splitting I mentioned above. 
So I made experiments with touching things to the output connectors, and even an insulated screwdriver triggered the alarm, which was only touched to the Limit Z, X or Y output. If I was touching the ground output, however, nothing happened. 
And more proof of interference is when I held my soldering iron (pistol like, bent copper wire, heavy metal stuff in it) and switched it on, the alarm was immediately triggered. 

So I think that the GRBL issue might be closed, because interference hasn't got much  to do with it. 
Am I right? 
What should I do to avoid interference? Move the power supply further from the arduino or place a metal shield between them?

electrokean, 2017-05-26:
Yeah, I was just about to suggest it sounds like a noise issue causing constant interrupts
Add some capacitors on each limit input pin (and probe input if you plan to use it)
See https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches for more detail

norbigazda98, 2017-06-07:
It's been some time since I was here, I didn't really have time to spend some with it.
Today I tried those connections and it works!
Well, amost.

So the limits are normally open (so normally no current goes through the limit switches). Hard limits and homing cycle is enabled. Limit switcth delay is 50 ms and limit pins are not reversed (so the setting is 0, also, tried 1, but that's not better), homing dir invert mask is also 0. 
I made the wiring you showed me above this message and sometimes it works. The change is that when hard limits are enabled and the stepper drivers' PSU is on, the SourceRabbit program doesn't freeze and I can move the motors pretty well. 

When I issue the $H command, again, the program doesn't freeze as before, it searches the upper Z limit switch well, and then it starts to search for the X and the Y limit switches at the same time, and after a few seconds, the software freezes, the motors stop and then the program says something like "error response 9" (sorry, I don't remember it exactly), and then an empty line appears and then it says that the connection has been estabilished. The homing cycle is not finished at this time, but when I kill alarm, I can move the motors freely. 

Can you help me a little bit more? :)

electrokean, 2017-06-07:
First, if you're seeing a numbered error or alarm code, then I believe you are using Grbl 1.x, and this repository is for the older 0.9 versions.  If using 1.1 you should open an issue at https://github.com/gnea/grbl/issues/

Second, I presume this is alarm code 9, which is a homing failure
See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message

Not sure what is happening, but have you got any filtering on your limit inputs to remove noise?

chamnit, 2017-06-07:
@norbigazda98 : I'm not familiar with SourceRabbit Gcode Sender. It could be a bug on their end. It's always good to try other Grbl GUIs to compare behavior. This will help identify where the problem is coming from, either Grbl or the GUI. 

Also, given that it says a connection has been established, it leads me to think you have a USB connectivity problem as well. Use a shielded USB cable with a ferrite core on it. This will help.

kenansatir, 2017-11-02:
Please use RFI, EMI, EMCi Noise Filters on limit switch cables (near to board) like :
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20171102041903&SearchText=+RFI+EMI+EMC+Noise+Filter

But try to use big ones... about 2cm diameter.... 

Also connect your boards GND pin to ground...

Best regards