grbl/grbl #1237: Arduino Libraries

algo1980, 2017-05-29:
Hi, why cant i use arduino library with grbl, do i have to make any changes to make file? 

i tried to add #define for arduino too but invain

X3msnake, 2017-05-29:
Grbl only uses the arduino IDE to make it easy for people to upload the
code.

Grbl code is highly optimized and is using almost all the available space
in the memory. No space left for libraries.

X3msnake, 2017-05-29:
https://github.com/grbl/grbl/issues/1231

No dia segunda-feira, 29 de maio de 2017, Vinicius Silva <x3msnake@gmail.com>

algo1980, 2017-05-29:
yes i read that earlier. but how can i compile this then, can i achieve this using atmel studio 7?

chamnit, 2017-05-30:
Arduino libraries have a ton of dependencies. Wading through that will require you to fully understand the Arduino source code in its entirety. Good luck with that.

closed 2017-05-30 by chamnit