grbl/grbl #1238: Atmel Studio

algo1980, 2017-05-29:
Is it possible to compile grbl using atmel studio7?

X3msnake, 2017-05-29:
Apparently yes

https://github.com/grbl/grbl/issues/424

Google is your friend. Use it ;)

chamnit, 2017-05-30:
Grbl does not support Atmel Studio, but it has been reported to work. Grbl only supports Makefiles or the Arduino IDE.

closed 2017-05-30 by chamnit

LETARTARE, 2017-05-30:
Hello,
for information : I use 'Code::Blocks' to compile and load Grbl (without using the 'Makefile').
Have a good day.

algo1980, 2017-05-30:
hi cham i am slightly confused here. as you mentioned GRBL only supports makefiles or the arduino ide but previously you said arduino ide cant compile grbl. what is the best method to compile. 

I have downloaded codeblocks do i have to download a seperate compiler for codeblocks too?

chamnit, 2017-05-30:
@algo1980 : I've never said Grbl can't be compiled via the Arduino IDE. It can, but if you try to add anything other than the grbl.h include, it'll bring in Arduino code and break. There is a very thorough wiki page on how to compile and flash Grbl through the IDE and how to use/setup the Makefile.

algo1980, 2017-05-30:
Thanks chammit ill read through that. You are a diamond.