grbl/grbl #1239: delay

algo1980, 2017-06-01:
hi i need to add some delay for lcd initialization. But it gives an error on 
#include <util/delay.h>

can you suggest how i can solve this.

cri-s, 2017-06-01:
If you relay on _delay_ms() you can do this:

#include <nuts_bolts.h> // if not already included
#define _delay_ms(x) delay_ms(x)
#define _delay_us(x) delay_us(x)

Otherwise use delay_ms and delay_us directly

algo1980, 2017-06-01:
thanks cri-s

algo1980, 2017-06-01:
tried it but it says multiple libraries found for nuts_bolts.h

cri-s, 2017-06-01:
can it be that you have included several version of grbl as arduino lib .

algo1980, 2017-06-01:
No i dont think so cri-s. i have only one version.

algo1980, 2017-06-01:
fixed it :)

X3msnake, 2017-06-01:
what was it?

algo1980, 2017-06-01:
it was funny i could not understand it either. I just restarter AS7

closed 2017-06-01 by chamnit

algo1980, 2017-06-01:
chamm why you close all my issues? :)

X3msnake, 2017-06-01:
Because it's Solved :)

gerritv, 2017-06-01:
And because you keep asking basic compile questions that have 0, zip, nothing to do with Grbl. The Arduino forums will provide better help for learning. Also you never seem able to provide even an error message, meaning the rest of us volunteers need to keep playing 20-questions with you.
Lastly you keep asking questions on a deprecated version of Grbl.

algo1980, 2017-06-06:
There is nothing wrong in asking a basic question. if you do not know the answer you can ignore it.:)

X3msnake, 2017-06-06:
@Tracy

Tracy, Gerrit has a point here.

It is not the basic question in itself it is the fact that it is not
related to supported methods of GRBL :)

algo1980, 2017-06-06:
@X3msnake 
Hi, i believe that extending GRBL support to next level will boost GRBL dev more. More DEVS means more development. :) But its the GRBL owner who have to decide. I can only put up my suggestion.

X3msnake, 2017-06-06:
The next release of grbl will have hardware abstraction that will allow
grbl to be developed in any hardware.

chamnit, 2017-06-06:
@algo1980 : Or, you simply don't have the time to answer and/or it has been answered several times in the past. Keep in mind that typing a response can take a couple of minutes onward to over ten minutes. That is a lot of (free) time you asking someone to spend on your question, when you can usually google for the answer yourself in that same period of time. Another benefit of doing this is that you can also learn other things along the way. I do this all of the time and credit stackexchange for a lot of the coding techniques inside Grbl. Most of which I ran across when I wasn't looking for it explicitly.

X3msnake, 2017-06-06:
Yep, Serendipity ;)