grbl/grbl #1243: ALARM : Homing fail Grbl V0.9

mo7amedelhadi, 2017-06-04:
Hi ;

I have a problem with my grbl, i try to realize a 3d printer. The problem is when i run grbl and when i click $H the printer does not go to the beginnings of the axes x0 y0 z0. 

I am working with CNC Shield V3 & Raspberry Pi 

These are the settings.

$H
ALARM: Homing fail
Grbl 0.9j ['$' for help]
$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000001)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=20.000 (x, step/mm)
$101=20.000 (y, step/mm)
$102=20.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Someone can help me please ?

algo1980, 2017-06-04:
look at your interface carefully.

mo7amedelhadi, 2017-06-04:
@algo1980 i'm using Universal Gcode Sender as an interface, i don't think that the problem is from the interface

algo1980, 2017-06-04:
try clicking return to zero.

milezdr, 2017-06-04:
You cant use grbl for 3d printing because you have only 3 axis you don't have axis for extruder

mo7amedelhadi, 2017-06-04:
@milezdr i use what :( ?

mo7amedelhadi, 2017-06-04:
@algo1980 it works , but the $H doesn't working

algo1980, 2017-06-04:
read the code and edit it according to your needs. You will have to consider writing / modifying your code according to your need. Grbl is open source. Grbl works great as its a great piece of software. You can play with it for a couple of months to learn it. :+1:

vMeph, 2017-06-04:
@mo7amedelhadi  for 3d printing you use marlin software
https://github.com/MarlinFirmware/Marlin
but it uses the arduino mega and probably you will need a better controller shield like ramps 1.4 has exemple