grbl/grbl #1245: SPI and GRBL

Alexsem76, 2017-06-07:
Dear developers, developers and users of GRBL control, I ask you this question: arduino uno hardware spi has 74hc165 and 74hc595, nobody tried to use them to expand the possibilities, namely to deduce limits on them, control thereby freeing the pins for the 4- Rotary axis or spindle control? Sorry for my english if that did not say so. I work as an industrial automation engineer and I successfully use this method to expand the inputs of the outputs arduino. Maybe this will help. I'm not a C ++ programmer. It is difficult for me to realize.

bluebriant, 2017-06-07:
Interesting thought. My level of knowledge in this regard is low. Could
anyone explain and say if this is possible? Freeing up additional pins on
the Uno would be welcome.

vMeph, 2017-06-07:
I dont think that will be possible with arduino uno GRBL, sense there is no memory available, the litle memory that is avaliable is for fix possible bugs, this have been mention alot times

Alexsem76, 2017-06-07:
А можно поинтересоваться где это упомянуто хочу почитать

lalo-uy, 2017-06-07:
I believe that the UNO version has get as good as it can.
For a similar price tag there are ARM modules with much more
pins/ram//flash/Mhz
The evolution should go  that way.
My humble opinion.

chamnit, 2017-06-07:
SPI support is planned to be installed in the ARM version, which has been under development for over a couple of years and the initial release coming out soon. As it turned out, the new overrides in v1.1 fit on a 328p and was released for it until a few more things in the ARM version are worked out.

closed 2017-06-07 by chamnit

Alexsem76, 2017-06-07:
Thank you very much, I will wait for new version